nodemcu/nodemcu-firmware #960: on('receive', ...) support chunked responses

dvv, 2016-01-20:
The rationale is to allow for polling (Content-Type: text/event-stream)

I saw #590 kinda touched chunked responses but discussion led to nothing on is it ever possible in nodemcu?

TIA

devyte, 2016-02-04:
The TCP/IP layer may break a payload into many pieces, and exactly how many is completely dependent on what the traffic nodes do over the entire trajectory. Then, there's the fact that the two end-point devices could potentially be anything, like two serial ports (picture two uart devices connected each to an ESP, and the 2 ESPs talk to each over wifi). In such cases, the TCP/IP layer is just a tunnel, hence the timing and packet sizes could be different (think modbus).
What this means is that whether it makes sense or not to accumulate packets and call one or multiple on("receive) callbacks is very dependent on exactly what is being transferred. The overlying protocol also would mandate how packets are accumulated. In your case, I think you want to accumulate packets to assemble an http response, hence the http protocol imposes the behavior. I can think of other protocols over TCP/IP which would have different behavior.
Then there's the case of a multi-protocol server, which wouldn't work if packets were to get accumulated.
I think response chunks should get accumulated based on your specific needs on the lua side.

dvv, 2016-02-04:
Reading the subj I realised I was too brief to explain.

I really meant not portions of data at application level, but responses in  https://en.wikipedia.org/wiki/Chunked_transfer_encoding instead.
Consider the following code that subscribes to events of `Content-Type: text/event-stream` (e.g. http://www.w3schools.com/html/html5_serversentevents.asp):

```lua
    local fd = net.createConnection(net.TCP, 0)
    fd:connect(port, host)
    fd:on("connection", function()
      fd:send(
          ("GET %s/sub HTTP/1.1\r\nHost: %s\r\n\r\n"):format(uri, host),
          function()
            fd:on("receive", function(_, s) ----- <<<< here we get _raw_ CHUNKS
              if type(s) == "string" then
                -- NB: ad-hoc poorman chunked encoding parser
                for e in s:gmatch("\r\ndata: (.-)\r\n") do 
                  e = cjson.decode(e)
                  _handle(e)
                end
              end
            end)
          end
        )
    end) 
```

So the subject is how about parsing chunked encoding at `net` level and delivering decoded chunks to `on("receive", ...)`?

(Hope this time it's clearer)

closed 2017-03-09 by marcelstoer