nodemcu/nodemcu-firmware #961: Discussion: moving config directory

robertfoss, 2016-01-20:
A problem I had while writing the enduser_setup module was that it was not possible to enable to module and it's dependencies from user_modules.h, since a dependency is located in app/include/lwip/app/dhcpserver.h. Which means that defines in user_modules.h won't be available in dhcpserver.h.

And this results in that anyone compiling this module will have to dig around a bunch of source code to be able to correctly enable everything that is required for this module. I would Ideally like to avoid this somehow.

Does anyone have any thoughts on this issue? Are there better solutions than moving the config dir? Am I doing something wrong?

jmattsson, 2016-01-21:
Doesn't it merely need `USE_DNS` defined in `dhcpserver.h`, which is the default these days anyway?

robertfoss, 2016-01-21:
I wasn't aware of that. I'll just close this issue.

closed 2016-01-21 by robertfoss