nodemcu/nodemcu-firmware #962: About HTTP client

Hellsy22, 2016-01-20:
1. 1024b limit.
It should callback with a chunks of data - let the lua-user to make a decisions about dropping or collecting/parsing data. 
2. Fixed headers. Should use user-provided prior to default (User-Agent, Connection).
And fixed timeout isn't any good.
3. Close connection vs keep-alive. KA is efficient way 
4. Redirect (301, 302) can be processed at lua-side, but why?
5. DNS. No caching + "close connection" == a lot of requests.

Summary: It's very memory efficient vs lua httpclient and supports https (finally!). But lack of options and 1024b limit blocks any attempts to extend it (i.e. redirects, cookie parsing, getting multipart/form-data etc).

I can try to make a few changes if it helps.

TerryE, 2016-01-20:
Sorry, but I am a little old fashioned.  It would really help if you used complete sentences and less twitter-speak.  Please refer to a specific code baseline and make some coherent suggestions and / or a Pull Request (PR), so we can understand what you are talking about.

jmattsson, 2016-01-21:
Yes it is rather limited, and yes the points you raise are good^. We'd certainly welcome patches to improve the http client module.

^ the SDK did have an internal DNS cache last I looked

tlightsky, 2016-01-21:
I agree with @Hellsy22 , httpclient should have some feature like:`callback with a chunks of data`

jmattsson, 2016-01-21:
PRs most welcome! I won't have a chance to even look at such an enhancement for quite some time.

Hellsy22, 2016-01-21:
@TerryE, sorry for twitter-like messages, english isn't my native language, so... less text I'll write - less grammar mistakes will be done. But I'll try to describe my case.

LUA is slow and memory inefficient language, but it's an interpreter. That mean a live code upgrades for any IoT device without wiring them to the UART. Basicaly all my IoT devices looking for updates on boot and downloading it.

I started from standart http-client example and then come to my own LUA http-client library. It have only two problems: memory and https. Even after I split the code to 6 modules (for unloading) it takes up to 17k memory while working. And sometimes little amount if memory leaks after timeouted requests (I checked _G table - it was ok, so, I suppose, It's on firmware side).

Second problem was https. Some of my IoT devices saving data to google docs and it accept only https requests. I solved problem by proxing requests thru pushingbox.com (it accept http and can send https).

Native http/https client can solve all the problems mentioned above. So, I grab it for testing and... it wasn't able to retrieve any updates - 1k buffer limit.

I can change BUFFER_SIZE_MAX manualy, but it's doesnt looks like good long-term solution. And adding the callback with chunks can conflict with original module purposes - looks like it has been designed to retrieve short datablocks or commands.

So, before doing anything I decided to ask what to do. May be current http client is a draft and contributors already have the plan how to improve it?

Hellsy22, 2016-01-21:
@jmattsson, about dns-caching.

http.get("http://api.ipify.org", NULL, function(status, r) print(status, r) end);
run it manualy 7 times

DNSmasq log:
```
2016.01.21 11:31:07 query[A] api.ipify.org from 10.101.10.39
2016.01.21 11:31:10 query[A] api.ipify.org from 10.101.10.39
2016.01.21 11:31:11 query[A] api.ipify.org from 10.101.10.39
2016.01.21 11:31:11 query[A] api.ipify.org from 10.101.10.39
2016.01.21 11:31:12 query[A] api.ipify.org from 10.101.10.39
2016.01.21 11:31:12 query[A] api.ipify.org from 10.101.10.39
2016.01.21 11:31:13 query[A] api.ipify.org from 10.101.10.39
```

Btw, http**s** issue:
```lua
> http.get("https://api.ipify.org", NULL, function(status, r) print(status, r) end);
> -1	nil
```

Returns -1 while not breaking 1k limit:

wget --save-headers -qO- https://api.ipify.org
```
HTTP/1.1 200 OK
Server: Cowboy
Connection: keep-alive
Content-Type: text/plain
Date: Thu, 21 Jan 2016 08:35:34 GMT
Content-Length: 14
Via: 1.1 vegur

81.4.*.*
```

tlightsky, 2016-01-21:
@Hellsy22 where is your own LUA http-client library,can you share it?

Hellsy22, 2016-01-21:
@tlightsky I submit one more ussue https://github.com/nodemcu/nodemcu-firmware/issues/964, blocking my lib from work in current firmware version.

If it not be fixed, then I'll do workaround and put lib to github.

marcelstoer, 2016-02-26:
@Hellsy22 Thanks a lot for testing and for your feedback. However, there are way too many issues in the initial description for this to be tracked sensibly here on GitHub. Please create individual bug reports or feature requests so that each of them can be addressed separately (they might all have different resolutions). Closing this one.

closed 2016-02-26 by marcelstoer