nodemcu/nodemcu-firmware #964: socket:send problem

Hellsy22, 2016-01-21:
Previously: https://github.com/nodemcu/nodemcu-firmware/issues/913
I was wrong, it's not ```on:receive``` problem - it's ```on:send``` problem.

Bug example:
```lua
sk=net.createConnection(net.TCP, 0)
sk:on("connection", function (sck)
	sck:send("GET / HTTP/1.1\r\nHost: api.ipify.org\r\nConnection: close\r\nAccept: */*\r\n")
	sck:send("\r\n")
end)

sk:on("disconnection", function(sck) print "Disconnected\n" end);
sk:on("receive", function(sck, c)
	print("got: " .. c)
end)

sk:connect(80, "54.225.138.188")
```

if I change 
```lua
sck:send("GET / HTTP/1.1\r\nHost: api.ipify.org\r\nConnection: close\r\nAccept: */*\r\n")
sck:send("\r\n")
```
to 
```lua
sck:send("GET / HTTP/1.1\r\nHost: api.ipify.org\r\nConnection: close\r\nAccept: */*\r\n\r\n")
```

All be ok. On older (sep 2015) firmware both cases was ok.

I checked more:
```lua
sk=net.createConnection(net.TCP, 0)
sk:on("connection", function (sck)
	sck:send("first\r\n");
	sck:send("second\r\n")
	sck:send("third\r\n")
end)

sk:on("disconnection", function(sck) print "Disconnected\n" end);
sk:on("receive", function(sck, c)
	print("got: " .. c)
end)

sk:connect(8882, "10.101.10.1")
```

Result:
```
# nc -l -p 8882
nc: using stream socket
first
```

All consequent send was ignored.

pjsg, 2016-01-21:
Yes -- it turns out that with the later SDKs from espressif, you can only have *one* socket::send outstanding at once. This is *really* annoying. I worked around this by using coroutines. In my case, I was making an httpserver work, but the idea is the same.

You can take a look at https://github.com/marcoskirsch/nodemcu-httpserver for the stuff that I did (look in the commit log).

devsaurus, 2016-01-21:
This characteristic of the SDK is covered in the FAQ at https://nodemcu.readthedocs.org/en/dev/en/faq/#how-is-coding-for-the-esp8266-different-to-standard-lua Also see #730 for technical details.
Will be closed due to reasons stated in #719.

closed 2016-01-21 by devsaurus

Hellsy22, 2016-01-21:
@pjsg, @devsaurus Thanks for the clarification.