nodemcu/nodemcu-firmware #966: How to handle backward compatibility

Alkorin, 2016-01-23:
I open the issue to continue the discution on #951

The merge breaks backward compatibility. Syntax was
```
ws2812.writegrb(pin, string)
```

The new code ignores "pin" and uses only GPIO2.
- Do I remove the arg and update the doc ?
- Do I keed the arg and add an `if (pin != GPIO2) { print "Only GPIO2 is supported"; }`
- Do I reintroduce the old code to handle other pins but add a big warning in the documentation "If you use something else than GPIO2, happy tree friends are coming" ?

How do you handle such case in NodeMCU modules ?

skycoders, 2016-01-23:
another thing is the influence on the blue LED on ESP12E boards which is also controlled through GPIO02. It should be noted that it cannot be used any more.
My favorite would be to let people chose. There might be the case that no WiFi is needed, and that would leave them with more choices on how to use their PINs.

devsaurus, 2016-01-23:
I tend to vote for **backwards compatibility** and to revive the bitbanging code asap. But we'd serve two different masters again - the Lord of Stability and Count Lockula :grin:.
How about a global setting for the firmware: safe mode vs unsafe mode. Safe mode would be default after startup and the user is free to enter unsafe mode with let's say `node.unsafe_mode()`. Modules like `ws2812` would block all unsafe operations in safe mode (read: allow only GPIO2) while they're inviting happy tree friends for tea time in unsafe mode.

API doc of `node.unsafe_mode()` shows the SDK concepts of ISR blocking, callbacks etc. Links to FAQ and so on. While the specific module functions just tell that they enforce restrictions in safe mode. Avoids to scatter the same info all over the doc tree.

matgoebl, 2016-01-23:
I'd vote for `if(pin == gpio2) { uart } else { bitbanging }` plus warning in the docs.
There might be users out there with existing set-ups. This would be fully backward compatible.

I also noted, that the very first `ws2812.writergb()` after reboot does not work (initialization problem?).
So in order to clear all LEDs I need two lines:
`ws2812.writergb(rgb_pin,string.char(0)) -- this is because the first write does not clear led 1
ws2812.writergb(rgb_pin,string.char(0):rep(3*numleds))`

TerryE, 2016-01-23:
@devsaurus, the main issue with backwards compatibility is that the old version basically crashed regularly as it severely broken the Espressif guidelines on masking interrupts and use of Flash ROM for interrupt masked code.  As an alternative to this merge, why don't we just withdraw support for WS2812s entirely as this is the only approach that I know works.

Does anyone who is actually _using_ the WS2812 module want to use the old version?  If so we need backwards compatibility. If not, then let us just retire the old version.  
I am confused by your concept of unsafe mode as safe mode seems to equal break Expressif guidelines and crash, whilst unsafe mode equals comply with Expressif guidelines and work stabily -- or am I just missing something entirely here?

devsaurus, 2016-01-23:
> If so we need backwards compatibility. If not, then let us just retire the old version.

Fully agreed. Why keep it if nobody is shedding any tears over the other pins.

Alkorin, 2016-01-24:
@matgoebl I just pushed a fix on my ws2812-uart branch : https://github.com/Alkorin/nodemcu-firmware/commit/ad58f4557e1c91177d909e0d1d4c7ee6386f81a8

Bug was there : [image]
GPIO2 is 'HIGH' at reboot and the ws2812 chip behave like if we have sent 25 bits with the first at 1.

I've pulled it down, added a 10us delay (reset) and it works now for me. Is it good for you too @matgoebl ?

[image]

devsaurus, 2016-01-24:
I'll come over here :smiley:
The issue is gone for me as well - the first `ws2812.writergb()` after boot is ok now.

matgoebl, 2016-01-25:
@Alkorin thanx again for your fix: the very first `ws2812.writergb()` after boot now works here as well :+1: 

Just for curiosity I did `gpio.write(4,gpio.LOW)` after the first `ws2812.writergb()` - until the next reboot I was unable to recover ws2812 with any subsequent `ws2812.writergb()` (because `isUartInited` is already `1`).
We should at least document that.

Alkorin, 2016-01-25:
Yes... I wait until a collegial decision about the need of backward compatibility. When everyone will be happy (I hope...) with the decision, I'll clean all the code, update the doc, etc... :)

jmattsson, 2016-01-27:
Caveat: I don't know enough about the ws2812 stuff, so feel free to ignore me.

Normally I'd be shouting "backwards compatibility", but if the old/current driver is highly prone to crashing itself, the WiFi stack or the entire device, then I'd definitely lean in the direction of retiring the old version with prominient notice in the docs.

marcelstoer, 2016-01-27:
+1 from me for dumping backwards compatibility in this case

TerryE, 2016-01-27:
@Alkorin Thomas, when you fixed all these feedback issues, just reissue the PR

marcelstoer, 2016-02-09:
@Alkorin any update on this? We seem to more or less agree that we don't need backwards compatibility in this case. So, I'd like to close #970 (no need to revert) but you should create a PR to update the docs so they're in line with the current implementation.

Alkorin, 2016-02-09:
Hello, I'm in holidays this week, so I'll make soon a new PR with the new implem and doc updated ;)

marcelstoer, 2016-03-01:
#970 has since been closed and the revert branch was removed.

acoulson2000, 2016-03-17:
I don't entirely understand the issue here, however I am using three pins. I did this because I'm writing to a large number of LED's (350) and doing this as one string was causing reboots, presumably because the watchdog wasn't getting petted, although admittedly, this could have been my loops taking too long to build the string. I could use some guidance if I'm doing something bad, but otherwise, I vote for retaining multi-pin support, as it seemed to be the only way I could get it to stay running consistently.

TerryE, 2016-03-17:
@acoulson2000 Andy, the new driver design allows you to drive that number of LEDs on one string stablely without screwing up the network stack. Isn't it perhaps better to try the new driver with one string first? To me your argument is "I had to botch my app with the old driver to get it to work, so let's keep the old driver".

acoulson2000, 2016-03-17:
@TerryE, Yes, I agree. Now that I've looked a the referenced issue #970, that certainly sounds positive, and would probably fix my issue. I think I'll see if I can try it out this morning. The downside is, I do have a hardware build based on multiple pins, but I'll gladly trade stability for some breadboard rework.

TerryE, 2016-03-17:
Yup being able to use the WiFi solidly when using the WS2812 string is pretty desirable, IMO.

acoulson2000, 2016-03-17:
So, is the UART change currently on Dev or Master?

TerryE, 2016-03-17:
`dev` but see #1146.

acoulson2000, 2016-03-18:
Got it. Seems odd that different issues aren't just corrected on their own branch...or on branches based on some release calendar.. and merged in on a regular basis, bu, not being a contributor, I'm certainly in no position to criticize.

One last newbie-ish question, probably for @Alkorin, would it be possible and/or make any sense to also support GPIO15/TXD2 as an alternative supported pin? Just thinking it could well be possible GPIO2 is tied up for other purposes...and I do think it could be nice to be able to have two different "banks" or strings of WS2812's that can be updated independently of each other.

Alkorin, 2016-03-26:
@acoulson2000 according to @vowstar "D7 and D8 can configured as an UART, but the original UART will disabled when you do it. You can't use UART0 and 'UART2' at the same time." (http://bbs.nodemcu.com/t/uart2-exists-in-nodemcu-esp8266/162)

As UART0 is used by lua's interpreter i/o ... not sure it's a good idea.

acoulson2000, 2016-03-27:
Ahh..good to know. It's certainly not worth sacrificing UART0.

marcelstoer, 2016-04-10:
> Hello, I'm in holidays this week, so I'll make soon a new PR with the new implem and doc updated ;)

@Alkorin -> bump

cogitatio, 2016-04-11:
Oh boy. That this only works with GPIO2 is an essential information, that took me 2 days to find out. 
I created a PR for a documentation fix.
https://github.com/nodemcu/nodemcu-firmware/pull/1235

marcelstoer, 2016-04-24:
Even though this one has been on the list for #1146 from the beginning I think we should stop waiting for a fix and cross it off that list.

Alkorin, 2016-04-24:
See #1263

closed 2016-05-09 by marcelstoer