nodemcu/nodemcu-firmware #967: Build of dev branch fails due to sntp module

skycoders, 2016-01-23:
I build the firmware through the docker image. I just ran into an issue that it would not build with an "implicit declaration of system_get_time() method" error. I fixed it by  
#include "user_interface.h" 
in sntp.c file. As also the cloud build of DEV does not seem to work (began, but did not finish even after half an hour) it might have the same issue?

devsaurus, 2016-01-23:
I'll have a look - thanks for reporting.

closed 2016-01-24 by marcelstoer

marcelstoer, 2016-01-24:
> As also the cloud build of DEV does not seem to work (began, but did not finish even after half an hour)

This sent a shiver down my spin when I read it and I rushed to analyze the build logs. What I found confirmed my creeping feeling. There were dozens of failed builds that I wasn't aware of.

The system is configured to notify me by email if a build fails and it had done that reliably for months. Due to a misconfiguration at my ISP a (long?) while back those emails had all been held back in a mailbox I never checked - until I found out today. I also found a huge pile of messages from users in that mailbox (thank-you notes, NodeMCU support requests, could builder feature requests, etc.). So, I sent out lots and lots of mails today.
AND to make matters worse due to a f*** up *I'm* responsible for the affected users probably weren't notified either. @skycoders Can you confirm this? It surprises me though that none of those affected users sent an email to my primary mailbox i.e. the one listed at http://nodemcu-build.com/faq.php#get-help.

The cloud builder used to be a pretty good first early-warning system for bugs in the firmware. I'm sorry I let people down :disappointed:.

skycoders, 2016-01-24:
I did not get an email about the failure. Thats why I added it to the ticket that something seems weired. It worked totally reliable so far, and took always around 3 minutes to produce the firmware. I was wondering if my build fails because of local issues, so I tried your cloud build and it failed as well. 
To add one more message, I really appreciate that you have created that service, as it really makes life easier for first-starters :)
And sorry to mess up your Sunday with all those answer emails you had to write now...