nodemcu/nodemcu-firmware #976: Reset loop when using #define DEVELOP_VERSION

jfollas, 2016-01-26:
With the latest Dev branch (SDK 1.5.1) and DEVELOP_VERSION defined, I get a wicked reset loop. 

Modules: ADC, BIT, CJSON, CRYPTO, ENDUSER_SETUP, FILE, GPIO, HTTP, MQTT, NET, NODE, RTCTIME, SNTP, SPI, TMR, WIFI.

Can anyone else reproduce? I think this is a bit beyond my troubleshooting abilities.

```
Smode : sta(5c:cf:7f:06:28:53)
add if0

_flash_used_end:4027dacf
fs.start:80000,max:37c000
mount res: 0
Task task_lua started.
SIG_LUA received.
Fatal exception 28(LoadProhibitedCause):
epc1=0x4000bf80, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0001f700, depc=0x00000000
¢@*r…ÄP±µ±Kít¥dr×-@éù.©Ñª¡Kl8a êÉ*±T±±±©©ÄP‹®±@íù*±…¢ünzAl )ém~±V%!ù(P±+tÇ,±@j|µ‹.@íù*±‰ønzAl )ém~±r[!ù(P±+tÇ,±@j|±@íù*±‰ˆrñ(‰8xºvA,P%ár§(±¹
Smode : sta(5c:cf:7f:06:28:53)
add if0

_flash_used_end:4027dacf
fs.start:80000,max:37c000
mount res: 0
Task task_lua started.
SIG_LUA received.
Fatal exception 28(LoadProhibitedCause):
epc1=0x4000bf80, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0001f700, depc=0x00000000
```

jfollas, 2016-01-31:
After further investigation, this seems to be related to the size of the 0x10000 bin. For example, you can see here that SPIFFS starts at 0x80000. It seems that having firmware that crosses over 0x7c000 on the flash results in this reset upon boot, and with the number of modules plus DEVELOP_VERSION set, this build did just that.

(This is on a -12E module with 4MB flash, so there's plenty of room for storage)

Is there a max size for firmware? Or can someone explain the underlying mechanics of the compiler/linker/runtime/etc that is causing this?

pjsg, 2016-01-31:
Can you upload your map file? I want to see which function is faulting....

jmattsson, 2016-02-01:
@pjsg No mapfile needed, that's the ROM'd strlen() attempting to read from 0x1f700, which is in the unused/protected region of memory (and hence hitting the LoadProhibited exception).

jfollas, 2016-02-07:
To add more to this issue: I recently tried to recreate by adding a lot of modules so that the SPIFFS had to start at 0x80000 (was hoping to get the same Panic-Reset so that I could check the map file or provide further details). This time, however, I just got a bunch of noise spitting out of the UART (noise meaning the same character repeating forever).

In my observation, there's something bad about having a firmware image that exceeds a certain size and DEVELOP_VERSION is defined. If I only include one or two modules so that the firmware size is smaller, everything works as expected.

My toolchain is the esp-open-sdk (using 1.5.1) built with STANDALONE=y.

jfollas, 2016-02-08:
More research, stumbled upon the "2A" document: http://bbs.espressif.com/download/file.php?id=1074

On page 22/23, it talks about increasing the irom0_0_seg len. By default, it's 0x32000, but can go as high as 0xC0000 for a 4MB flash.

At first glance, this sounds very applicable to my issue.

ABonner, 2016-02-11:
I recently started doing dev and with the stock dev branch my `0x10000.bin` file was also flashing past 0x7c000 and got the same issue you did with resets and/or junk being output. Had to remove a lot of modules to get the size down, then everything was fine.

jfollas, 2016-02-13:
I'm going to close this issue and re-open under a different title. The root cause is not related to #DEVELOP_VERSION. Having that enabled just adds more code to the firmware, which increases the size. 

When the firmware gets beyond a certain size (seems to be 0x6c000 len for the 0x10000.bin), then the reset fails in various ways.

closed 2016-02-13 by jfollas