nodemcu/nodemcu-firmware #982: calculating sha1 hash of big files is not possible

kmpm, 2016-01-28:
I want to calculate checksum of big (> ~1460 bytes) files on the nodemcu but I can't find any way of doing it currently.

It seems as one would need some way of not reading the whole file into a string and then do checksum but read in blocks and update instead.

Examples.
* http://stackoverflow.com/questions/1869885/calculating-sha1-of-a-file
* http://luacrypto.luaforge.net/0.1/luacrypto.html ( look at d:update(string) )
* https://docs.python.org/2/library/hashlib.html

jmattsson, 2016-01-29:
I agree that would be nice to have. Alas I don't have time to add this support right now, but what I'd envisage is this sort of usage:
```lua
local sha = crypto.new_hash("SHA1")
sha.update("mystring1")
sha.update("more stuff...")
local digest = sha.finalize()
```
I believe we have all the internals needed, we "just" need to glue it together in the right way. Someone is most welcome to beat me to this work.

ABonner, 2016-01-29:
I'd be happy to try to implement this over the weekend. It looks like we would pass back the contents of the `SHA1_CTX` struct as userdata to persist the state across calls.

jmattsson, 2016-01-29:
I'd recommend `digest_mech_info_t` (or suitable subset thereof) plus the working buffer for the userdata. That way we'll automatically get the support for MD5 and SHA2 and/or whatever else has been enabled. Have a look at app/crypto/crypto.{h,c} for inspiration. :)

ABonner, 2016-01-29:
Thanks for the tip, I was looking at the `crypto_sha1` function instead since the topic mentioned SHA1. I'll look at the more generic way of doing it.

jmattsson, 2016-01-29:
Great, let me know if you have any questions once you dig in. Oh, and do the work off the `dev` branch please, it'll save you hassle once you want to raise the PR. And Marcel would probably like me to point you to [CONTRIBUTING.md](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) for other useful info too.

cheloftus, 2016-01-30:
Would it be beneficial to have the file IO handled outside of the Lua world? Seems like it might be more efficient for the file hashing case. Perhaps provide a crypto export that specifically hashes files?
I was envisaging something like:

```lua
local digest = crypto.fhash("SHA1", "myfile.ext")
```

I don't dispute that the flexibility of exposing functions like `new_hash`, `update`, `finalize` would be very useful in a more general sense though.

cheloftus, 2016-01-30:
Don't want to step on your toes @ABonner, but I've got a working example of crypto.fhash as per my comment. Tested on hardware and it produces expected results. The commit in my fork is referenced above; I haven't submitted a PR.

ABonner, 2016-01-30:
Nice, @cheloftus ! Looks like a good implementation to me. No worries about stepping on the code I am writing.

kmpm, 2016-01-30:
@cheloftus, that looks perfekt for me.

cheloftus, 2016-01-30:
Thanks for the comments, I'll submit a PR then.

ABonner, 2016-02-10:
I'll be submitting a PR for the generic version of this later today.

TerryE, 2016-02-10:
Have you done any timing benhmarks? I am worried this will blow the Espressif 10-15 mSec max time for any individual task or callback out or the water.

You might need to make this an async multi-task approach using the new task inferface, #1023, and a callback to pick up processing when done.

ABonner, 2016-02-11:
@TerryE I hadn't but did some very simple timing on this with SHA256 (ran it a few more times than shown and results are within 10% or so). Not sure the best way to time things, but this simple approach seems to indicate that creation of hash structure is 1ms, operations on 900 byte strings are about 1.5ms, finalizing is 2.5ms.

Since it is likely that a user would be calling this within another task that is already receiving chunked data (file I/O or network packets), it seems like it is within reason to do it synchronously.

I would be happy to add a task-managed version too or convert over if that seems like the best option for some reason.

> n=tmr.now; a=n(); x = crypto.new_hash("SHA256"); b=n(); print ("Elapsed us = " .. b-a);
Elapsed us = 1059
> n=tmr.now; a=n(); x:update(bigstring); b=n(); print ("Elapsed us = " .. b-a); -- 900 byte string
Elapsed us = 1597
> n=tmr.now; a=n(); hashstr = x:finalize(); b=n(); print ("Elapsed us = " .. b-a);
Elapsed us = 2397
> print (crypto.toHex(hashstr))
ad64fa338c0551e32845e61b9b4ce36aeb2fa67d2cdda54354d634837c50d298

TerryE, 2016-02-11:
What I was first asking what the equivalent `crypto.fhash()` time would be for files of a range of sizes, enough, say, to give an estimate linear formulae, so we could say that the SHA256 of a file of X bytes will take 15 msec and Y bytes for SHA1, and for files larger than this we should use a tmr.alarm "loop"  using the update approach.

ABonner, 2016-02-11:
Here is timing for a simple Lua implementation of the `crypto.fhash()` function.

Chunksize is how many bytes are read/processed per loop, FileLen is the total file size, Elapsed time is in usec.

**Summary:** For SHA256/SHA1, about 5000 byte file is the max before you hit 15ms. For MD5 you can get through a 10000 byte file within 15ms.

	Hash Algo       Chunksize       FileLen ElapsedTime
	SHA256          500             100     5077
	SHA256          500             500     5276
	SHA256          500             2000    9632
	SHA256          500             1000    6739
	SHA256          500             5000    18191
	SHA256          500             10000   32315
	SHA256          1000            100     4839
	SHA256          1000            500     5273
	SHA256          1000            2000    8047
	SHA256          1000            1000    5944
	SHA256          1000            5000    14046
	SHA256          1000            10000   23943
	SHA1            500             100     4180
	SHA1            500             500     4374
	SHA1            500             2000    7517
	SHA1            500             1000    5444
	SHA1            500             5000    13804
	SHA1            500             10000   24097
	SHA1            1000            100     4092
	SHA1            1000            500     4373
	SHA1            1000            2000    6339
	SHA1            1000            1000    4857
	SHA1            1000            5000    10626
	SHA1            1000            10000   17686
	MD5             500             100     4020
	MD5             500             500     4256
	MD5             500             2000    7114
	MD5             500             1000    5231
	MD5             500             5000    12892
	MD5             500             10000   22393
	MD5             1000            100     4020
	MD5             1000            500     4255
	MD5             1000            2000    5938
	MD5             1000            1000    4646
	MD5             1000            5000    9730
	MD5             1000            10000   15991

TerryE, 2016-02-11:
I've plotted this below, to make easier to see.  I suspect that a 1024 block size would be better, but MD5 and SHA1 are the fastest (since these run out of ROM), but a file size of 10Kb is bout the largest that we can do within the 15m mSec advised limit.
[image]

ABonner, 2016-02-11:
Nice graph, thanks!

1024 byte chunks have similar performance characteristics (see below), actually seems slower for some odd reason on files that are large enough to benefit.

I am good with recommending around 10000 bytes is the max that should be processed at one time (anything beyond should chunk the source up and use callbacks separate from the hash functions). Should I add a commit with documentation updates?

    Hash Algo       Chunksize       FileLen ElapsedTime
    SHA256          1024            100     5074
    SHA256          1024            500     5265
    SHA256          1024            2000    8919
    SHA256          1024            1000    5949
    SHA256          1024            5000    17796
    SHA256          1024            10000   32529
    MD5             1024            100     4109
    MD5             1024            500     4241
    MD5             1024            2000    6715
    MD5             1024            1000    4635
    MD5             1024            5000    12938
    MD5             1024            10000   23274

TerryE, 2016-02-12:
I'll check the source code over the weekend.  I suspect that the physical block size is a multiple of 512 byte, but this might include some bookkeeping pointers so the application logical blocksize _might_ be smaller.  I think I'll do a similar benchmark of pure read performance and write performance. :smile:

closed 2019-06-21 by stale[bot]