nodemcu/nodemcu-firmware #986: MQTT: Subscribing to multiple topics at once loops forever

devsaurus, 2016-01-28:
I came across this during work on #985: If 3 or more topics are to be subscribed using the "tabular" variant of `m:subscribe()` then the process loops forever.

This works as expected:
```lua
> m = mqtt.Client("esp8266", 60, "", "")                                                                                 
> =m:connect(server_ip, 1883, 0, 1, function() print("connected") end)
true                                                                                                                        
> connected
=m:subscribe("test1", 1, function() print("subscribed test1") end)
true
> subscribed test1
=m:subscribe("test2", 1, function() print("subscribed test2") end)
true
> subscribed test2
=m:subscribe("test3", 1, function() print("subscribed test3") end)
true
> subscribed test3
=m:subscribe("test4", 1, function() print("subscribed test4") end)
true
> subscribed test4
=m:close()
true
```

While things begin to break with `m:subscribe()` for 3 topics:
```lua
> m = mqtt.Client("esp8266", 60, "", "")                                                                                 
> =m:connect(server_ip, 1883, 0, 1, function() print("connected") end)
true                                                                                                                        
> connected
=m:subscribe({tab2_1 = 1, tab2_2 = 1}, function() print("subscribed table with 2 elements") end)
true
> subscribed table with 2 elements
=m:subscribe({tab3_1 = 1, tab3_2 = 1, tab3_3 = 1}, function() print("subscribed table with 3 elements") end)
true
> =m:subscribe("test4", 1, function() print("subscribed test4") end)
false
> =m:close()
true
```

The callback for `print("subscribed table with 3 elements")` is never triggered and subsequent `m:subscribe()` fail. Closing and setting up the client from scratch is the only way to resolve the situation.

Mosquitto log shows:
```
1454019172: New connection from client_ip on port 1883.
1454019172: New client connected from client_ip as esp8266 (c1, k60).
1454019175: esp8266 1 tab2_2
1454019175: esp8266 1 tab2_1
1454019203: esp8266 1 tab3_2
1454019203: esp8266 1 tab3_1
1454019203: esp8266 1 tab3_3
1454019204: esp8266 1 tab3_2
1454019204: esp8266 1 tab3_1
1454019204: esp8266 1 tab3_3
1454019205: esp8266 1 tab3_2
1454019205: esp8266 1 tab3_1
1454019205: esp8266 1 tab3_3
1454019206: esp8266 1 tab3_2
1454019206: esp8266 1 tab3_1
1454019206: esp8266 1 tab3_3
1454019207: Client esp8266 disconnected.
```

tab2_* subscriptions work fine while tab3_* subscriptions repeat endlessly until `m:close()` is called.

Guess this worked at some point in the past as it's described in the README (on `master`).

pjsg, 2016-03-10:
From looking at the code, it would be a minor miracle if it ever worked.

Anyway, I have fixed it (and I tested with 10 subscriptions).... I'm pushing the changes to PR #1129 which is already open for related MQTT issues

TerryE, 2016-03-14:
@pjsg Philip, in this case do you want to close this issue?

pjsg, 2016-03-15:
I'll close it once the underlying PR is merged.....

pjsg, 2016-03-17:
The PR was merged and this should be resolved.

closed 2016-03-17 by pjsg