nodemcu/nodemcu-firmware #990: Docs for flashing custom build not in dev branch

ABonner, 2016-01-30:
I was getting going on some dev for NodeMCU and noticed that some of the instructions on building and flashing starting with the repo were present in the 1.4.0 master branch but not in the new docs in dev.

I can move over some information and expand on it a bit as I just went through the process, but wanted to check that I wasn't just missing it.

jmattsson, 2016-01-30:
More than what's available on the [building](http://nodemcu.readthedocs.org/en/dev/en/build/) and [flashing](http://nodemcu.readthedocs.org/en/dev/en/flash/) pages?

marcelstoer, 2016-01-30:
> the 1.4.0 master branch documentation

...is still there - in the `master` branch. What is it that you're missing? Even the wiki page has a [history](https://github.com/nodemcu/nodemcu-firmware/wiki/nodemcu_api_en/_history) which you can browse.

ABonner, 2016-01-30:
Sorry for not being clear. I meant the [Files to Burn to the Flash](https://github.com/nodemcu/nodemcu-firmware#files-to-burn-to-the-flash) section in particular was not on the [flashing](http://nodemcu.readthedocs.org/en/dev/en/flash/) page. I was proposing to move that over to the flashing page in the new documentation.

closed 2016-01-31 by marcelstoer

ABonner, 2016-02-01:
Looks great @marcelstoer !

commit 9e861313 (2016-01-31): Add chapter about firmware files and flash address

Had to slightly tweak the Markdown headers due to
https://github.com/mkdocs/mkdocs/issues/770 :(.

Fixes #990.