nodemcu/nodemcu-firmware #995: How to deploy firmware in mass?

jfollas, 2016-01-31:
I've been thinking about what it takes to deploy NodeMCU + scripts to a lot of identical devices with known flash memory sizes. Information seems to be spread around among different projects for the ESP, so having this all in one place would be useful for others, too.

Spiffy doesn't seem to produce compatible file system images these days, so what I'm thinking of instead is to just manually load scripts onto one device, and then dump the filesystem using esptool. Then I'll have three images to load: 0x00000.bin, 0x10000.bin, and my fs.bin (let's say it gets loaded to 0x7c000).

Questions about things that are still a bit vague to me:

1. If I only have 5 KB of scripts in SPIFFS when I do the dump, is it okay to just dump the bare minimum? Meaning, does SPIFFS start writing at the beginning of it's address space and the rest remains as 0xFF until actually needed?

2. Is there another .bin required to make my image complete? Init_data, perhaps? Is that at the end of the flash memory space? And how large? Is this the same for all devices (i.e., no device-unique info stored by default)?

3. What changes need to be done to keep SPIFFS from creating/formatting the filesystem on first boot, which would replace the fs.bin that was loaded? Or would having the init data solve that problem?

4. Since my hardware is well-known to me, should I use FLASH_4MB in user_config.h instead of FLASH_AUTOSIZE? Should I disable FLASH_SAFE_API? Any benefit to doing this?

jmattsson, 2016-02-01:
The issues list is probably not the right place for these questions, but:

0.5 - Try [spiffsimg](https://github.com/DiUS/spiffsimg) instead of spiffy?

1. I believe this is the case, yes.

2. Only if you need a customized init data block. NodeMCU comes with its own embedded copy which it applies if there is no existing init data block in flash. See the Espressif user guide for details on location. There is no device-identifying information, only configuration settings.

3. NodeMCU only creates/formats the SPIFFS volume if it doesn't already exist. If you write your fs.bin to the right place, NodeMCU/SPIFFS will detect it and use it.

4. I wouldn't bother. You would slightly reduce the code size, but the autosize functionality is pretty small anyway. Plus it's the option that has seen most of the testing :) Safe-mode effectively distrusts the size byte in ROM, allowing access to the full flash.

TerryE, 2016-02-01:
My comment would be that the ESP is a WiFi enabled chip.  I provision my "prod" devices by including a SPIFFS which includes a standard bootstrap tht uses UDP to interrogate a known UCP socket on a (RPi2) provisioning server.  You plug the chip in and it configures itself in a few seconds writing the app and config data for that module on its FS.  

However, at the moment we are not proposing to introduce or to enforce a standard provisioning model so this falls under #719 and I am closing this discussion.

jfollas, 2016-02-01:
No worries about closing the issue - it was more of a documentation request than anything.

closed 2016-02-01 by jfollas