nodemcu/nodemcu-firmware #996: NodeMCU speaks TLS 1.1 instead of TLS 1.2

trahloff, 2016-02-01:
Some rejected handshakes from a MQTT broker indicated that the TLS implementation is wrong in some way.
After a colleague of mine set up some test stands, doing several days of testing and analysing the results he comes to the conclusion that the NodeMCU speaks TLS 1.1. Let me show you the results:

**This is the NodeMCU client hello:**

0000   16 03 02 00 33 01 00 00 2f **_03 02_** 00 00 00 00 d0
0010   b1 a1 3a 07 1c 1b 3e f2 fc 03 91 d6 18 b5 ae 5d
0020   77 65 37 f5 07 10 45 d1 7e 1a ea 00 00 08 00 2f
0030   00 35 00 05 00 04 01 00

**The parsing of that:**

16 -	   SSL3_RT_HANDSHAKE (Is this right - ie a common SSL3 and TLS byte here?)
03 02 - 	   version : 3.2
00 33 - 	   length of record excluding 5 bytes of header
01 - 	   SSL3_MT_CLIENT_HELLO
00 00 2f - length of the following data
03 02 -    TLS v1.1
00 00 00 00 d0 b1 a1 3a 07 1c 1b 3e f2 fc 03 91d6 //-- ?.... - Random 16 bytes

**A proper TLS 1.2 client hello looks like this:**

0000   16 03 01 00 7e 01 00 00 7a **_03 03_** 56 ab 8e f3 59
0010   bb 39 71 b4 9f d1 4c bc 39 f3 37 ca 8c bc f8 f4
0020   d8 6f 40 4b cf a9 a4 48 62 c1 70 00 00 10 00 9c
0030   c0 2f c0 2b 00 a2 00 9e c0 31 c0 2d 00 ff 01 00
0040   00 41 00 0b 00 04 03 00 01 02 00 0a 00 08 00 06
0050   00 19 00 18 00 17 00 23 00 00 00 0d 00 20 00 1e
0060   06 01 06 02 06 03 05 01 05 02 05 03 04 01 04 02
0070   04 03 03 01 03 02 03 03 02 01 02 02 02 03 00 0f
0080   00 01 01
(This packet was captured from a browser connecting to the same test stand with TLS 1.2)

After enabling TLS 1.1 on the test stand the NodeMCU could publish without any problems.
This is based on reading the rfcs for tls 1.1 and 1.2.

jmattsson, 2016-02-01:
Interesting. Is this on master or dev?

trahloff, 2016-02-01:
Master

jmattsson, 2016-02-01:
Any chance you could test dev too? `dev` is on 1.5.1, whereas master is still 1.4.0

trahloff, 2016-02-01:
Thanks for the hint. I will test this.

trahloff, 2016-02-01:
Update: same problem with dev

jmattsson, 2016-02-01:
Right, sounds like we need to update the documentation then!

trahloff, 2016-02-01:
I spoke to Espressif, they said TLS 1.2 will come 2017...

marcelstoer, 2016-02-01:
@trahloff the reference in #539 concluded it must be TLS 1.2 based on the return values of calling https://www.howsmyssl.com/a/check. Did you run that check as well and if so did it return 1.2 or 1.1?

trahloff, 2016-02-01:
No but I wrote to Espressif (the "feedback" address) and they said:

> Currently our SDK don't support TLS1.2, we plan to  update it next year.

marcelstoer, 2016-02-06:
I updated the documentation accordingly.

Running the following code
```lua
http.get("https://www.howsmyssl.com/a/check", nil, function(code, data)
    if (code < 0) then
      print("HTTP request failed")
    else
      print(code, data)
    end
  end)
```
yields
```
{
	"given_cipher_suites": ["TLS_RSA_WITH_AES_128_CBC_SHA", "TLS_RSA_WITH_AES_256_CBC_SHA", "TLS_RSA_WITH_RC4_128_SHA", "TLS_RSA_WITH_RC4_128_MD5"],
	"ephemeral_keys_supported": false,
	"session_ticket_supported": false,
	"tls_compression_supported": false,
	"unknown_cipher_suite_supported": false,
	"beast_vuln": false,
	"able_to_detect_n_minus_one_splitting": false,
	"insecure_cipher_suites": {
		"TLS_RSA_WITH_RC4_128_MD5": ["uses RC4 which has insecure biases in its output"],
		"TLS_RSA_WITH_RC4_128_SHA": ["uses RC4 which has insecure biases in its output"]
	},
	"tls_version": "TLS 1.1",
	"rating": "Bad"
}
```
Quite frustrating to see the SDK our firmware is based upon rated as "Bad" :frowning:.

closed 2016-02-06 by marcelstoer

jmattsson, 2016-02-07:
!?!

How could your output looks so widely different to the one linked in #539? We're not building and using an old libssl.a are we?I thought even the possibility of doing that disappeared with the 1.4.0 upgrade?

marcelstoer, 2016-02-07:
> How could your output looks so widely different to the one linked in #539?

That's what bothers me as well. Whatever it is, the documentation is now in line with the code.

marcelstoer, 2016-02-08:
2nd answer to the same question

> How could your output looks so widely different to the one linked in #539?

Because we were fools...that original tweet doesn't mention ESP IoT SDK or NodeMCU at all, it only says "ESP8266's TLS". Bogus of course, as if the ESP hardware had anything to do with the TLS implementation of the *firmware* that runs on top of it. Yet, we concluded the author was talking about the "our" SDK.

Turns out he was not. [He was using ESP *FreeRTOS* SDK](https://twitter.com/TimMattison/status/696737109160366082).

stgoddv, 2017-10-12:
Hi! Any updates about this?
MQTT now is still running upon TLS 1.1?

Thanks!

commit e9cde6eb (2016-02-06): Correct secure option in MQTT documentation

Fixes #996