nodemcu/nodemcu-firmware #1006: Pulsen defined as uint8 is too short In  nodemcu-firmware/app/modules/rc.c

ftoure67, 2016-02-03:
The uint8 value is to short for the pulse len. Some switches require a 375 ms pulsen delay. The uint8 value can only handle delays up to 255.

nickandrew, 2016-02-04:
You simply cannot delay that long without yielding control to the SDK, and expect a stable device. The RC module is very bad.

TerryE, 2016-02-04:
:+1: It desperately needs a tmr-based version written.

nickandrew, 2016-02-04:
What kind of device must be attached to the output pin? Not a simple IR LED. IR needs to be modulated at (e.g.) 38000 Hz and so several (say 10) pulses are required for a single halfbit output.

ftoure67, 2016-02-04:
Hi Nick,

It's a Chacon RF433 Mhz transmitter. I have made it work previouly with
another MCU with the delay set to 375. I have checked the rc.c code. The
protocol 1 is perfect, except the delay that it limited to 255.

Regards,

Famory

ftoure67, 2016-02-04:
Hi Terry,

Yes. I will try with a tmr based version. I still think that a version
included in the firmware would do a better job even if the current version
of rc.c is not so simple to handle.

Regards,

Famory

nickandrew, 2016-02-04:
@ftoure67 I expect Lua code which used rc.c for short intervals could also use tmr for long interval delays like 375 ms. I don't understand enough about the protocol to know if the rc.c module itself can be easily modified to use a timer for all longer delays.

ftoure67, 2016-02-05:
Hi Nick,

I have implemented the lua code that works for my device, using the
tmr.delay. The correct value is 375 us (not ms). The Lua code for my device
is exactly the one currently implemented in rc.c. So there is no protocol
change required in the code. The only point is that the delay coded with an
uint8 cannot handle this 375 us delay.

Regards,

Famory

nickandrew, 2016-02-07:
How long does the total bit stream take, from beginning to end?

ftoure67, 2016-02-08:
Hi Nick,

The protocol for my device is using 12 "bits" (should be named waveforms,
since a bit is only for 0 and 1) plus a Sync at the end.
In rc.c, the code is only considering 0 and 1 bits coded as:
-- "0" Bit (1*HIGH, 3*LOW, 1*HIGH, 3*LOW)
--            _     _
-- Waveform: | |___| |___
--
-- "1" Bit (3*HIGH, 1*LOW, 3*HIGH, 1*LOW)
--            ___   ___
-- Waveform: |   |_|   |_
--
-- "S" Bit for synchronization (1*HIGH, 31*LOW)
--            _
-- Waveform: | |_______________________________

There is one waveform that is missing in rc.c. It's called "F" bit:
-- "F" Bit (1*HIGH, 3*LOW, 3*HIGH, 1*LOW)
--            _     ___
-- Waveform: | |___|   |_

At final, to put the (1, 1) light on, I have to send the following sequence
of waveforms:0FFF0FFFFFFFS.

Regards

Famory

TerryE, 2016-02-08:
@ftoure67 Github flags email responses as (proportional) text only so any fixed column text diagrams get screwed up, so its better posting these directly on Github through HTTPS.

nickandrew, 2016-02-10:
So counting @ftoure67 protocol, "0FFF0FFFFFFFS" is 128 x `pulseLen`. You want to set `pulseLen` to 375 µs, which means the total sequence length will be 48ms, which is well over the maximum time recommended by the SDK. Correct me if I'm wrong here @TerryE but your FAQ says 

> The latest SDK includes a caution that if any (callback) task runs for more than 10 mSec, then the Wifi and TCP stacks might fail,

NodeMCU should adopt some number, and say "this is the maximum amount of time we'll support bit-banging in one task". 10ms might be a bit dodgy; 5ms is safe; whatever the number is, 48ms is much larger.

So I am inclined to :-1: changing pulseLen to uint16 on this basis.

@ftoure67, am I correct on my calculations above? Does the protocol allow a delay after sending each bit?

On your description, sending just a "0" bit involves 375µs HIGH, then 1125µs LOW, then 375µs HIGH, then 1125µs LOW for a total of 3000µs or 3ms. After that bit, the CPU really needs to take a break and look after WiFi.

I'm also trying to understand how your protocol is usable with the "rc" module as it stands (ignoring the uint16 thing). To send "0FFF0FFFFFFFS" in "rc", the bit pattern passed to the module needs to be "000101010001010101010101" and the S bit is added automatically at the end. Just the S bit will be 12ms in duration.

nickandrew, 2016-02-10:
Note I've edited the title and comment to refer to the correct "pulseLen" rather than "pulsen".

ftoure67, 2016-02-10:
> So I am inclined to [image: :-1:] changing pulseLen to uint16 on this

Hi Nick,

Yes. Your analysis is correct. As the protocol is using a tristate instead
of simple bits:
- 0 should be translated as 00 in rc,
- F as 01
- 1 as 11
Doing so, with the correct delay set to 375, it should work with rc.c.
In the pure lua version, there is no need to call tmr.wdclr() when sending
a full code.

Regards,

Famory

snosrap, 2016-08-01:
I bumped into this issue as well.  Changing `pulseLen` from a `uint8_t` to a `uint16_t` solved the problem for me.  I think changing to an `uint16_t` is a good idea because

1. It will prevent silent integer overflow
2. The delay duration (~50ms) is acceptable according to the ESP8266 SDK docs
3. Using timers/callbacks won't work

### It will prevent silent integer overflow

By using an `uint8_t`, the current `rc.c` fails in an unpredictable way for `pulseLen`s over 255.  Rather than throwing an error, a debug message, or capping the value at 255, the code silently overflows the integer -- so my 412µs becomes 156µs (I may not have noticed this had it not been for the `NODE_ERR` debug messages in the rc.c module).  I think using an `uint8_t` to prevent excessively-long calls to `os_delay_us` is somewhat confusing for the developer.

### The delay duration (~50ms) is acceptable according to the ESP8266 SDK docs

In the [ESP8266 SDK docs](https://espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf) (Overview, Page 15):

> We suggest using a timer to check periodically, if users need to call function os_delay_us or while, or for in timer callback, please do not occupy CPU more than 15 ms.

But that's just talking about **callbacks**, right?  What about regular C function calls, as found in `rc.c`?:

> if interrupt is not disabled, it is suggested that CPU should not be occupied more than 500 ms.

500ms far exceeds the needs of the `rc.c` module to send a message (on the order of 50ms, as described above).

### Using timers/callbacks won't work

Additionally, per the docs, it might be impossible to re-write `rc.c` using microsecond timers (os_timer_arm_us, Page 18):

> The highest precision is 500 us.

Since most `pulseLen`s I've seen are in the 100-400µs range, I don't think timers are a feasible option.  Based on my tests, calls to `ets_timer_arm_new(&tmr, pulse_len, 0, 0);` for `pulse_len` values above 50-70µs were all over the map.

I'll readily admit, I'm new to ESP8266 programming -- but, based on my reading of the docs, it seems like a series of `os_delay_us`s adding up to about 50ms wouldn't cause problems as long as they weren't in a callback.

Changing `pulseLen` from a `uint8_t` to a `uint16_t` would make `rc.c` work with more devices and prevent the current state of silent integer overflow.

Any thoughts?

closed 2019-06-21 by stale[bot]