nodemcu/nodemcu-firmware #1007: Clarification please

fixingthingsguy, 2016-02-03:
In reading some of the issues, I note we are currently compatible with SDK 1.4.
I need to use the esp_init_data_default.bin to enable ADC(tout)(yes, sorry to bring that up again),so I'm wondering should I be using
        ESP8266 SDK（esp_iot_sdk_v1.4.0_15_09_18）.
        And  when I try to download it takes me to:
        [SDK Release] ESP8266_NONOS_SDK_V1.4.0_15_09_18
        And then I'm lost as the links are misleading.
Is there a location in nodemcu that I can pull in the right esp_init_data_default.bin compatible with the current dev node. Which may be the better way I think to assure consistency with frightanic builds.

Thank you for your time
Regards

devsaurus, 2016-02-03:
Please see [Upgrading from SDK 0.9.x Firmware](https://nodemcu.readthedocs.org/en/dev/en/flash/#upgrading-from-sdk-09x-firmware). It links the suitable SDK release and explains how to download the contained esp_init_data_default.bin.

nodemcu firmware doesn't re-distribute the file andI'm not aware of any plans to do so.

jmattsson, 2016-02-03:
Another option is to do a full chip-erase before flashing the new firmware (but of course, you'll lose all stored settings such as WiFi) . NodeMCU has its own built-in copy of the esp_init_data defaults which it applies on first boot if the init data block is missing on the device.

fixingthingsguy, 2016-02-04:
Thank you devsaurus and jmattsson.
I did the following, writing it here mostly if I need to revisit.
- downloaded the 1.4 sdk from https://nodemcu.readthedocs.org/en/dev/en/flash/#upgrading-from-sdk-09x-firmware
-  chip erase using blank.bin from above sdk;flashed at 0x7e00
-reflashed with my frightanic load based on a recent 1.4 base dev load
-applied the #911 prescription to change byte 107
-ADC was enabled
 For a semi-casual user, it looks complicated but it isn't and works fine. 
Once again, I appreciate your timely help
Regards

fixingthingsguy, 2016-02-04:
Thank you devsaurus and jmattsson.
I did the following, writing it here mostly if I need to revisit.
1. Downloaded the 1.4 sdk from https://nodemcu.readthedocs.org/en/dev/en/flash/#upgrading-from-sdk- 09x-firmware 
2.Chip erase using blank.bin from above sdk;flashed at 0x7e00
3. Reflashed with my frightanic load based on a recent 1.4 base dev load
4. Applied the #911 prescription to change byte 107 
5. ADC(TOUT)was enabled*
 For a semi-casual user, it looks complicated but it isn't and works fine.
 Once again, I appreciate your timely help
 Regards

*Expressif, if you read this, please consider reversing byte 107 to provision ADC(Tout). It distinguishes your device offering.   Arduino has provisioned ADCs. "Apparently", Raspberry PI does not provision.

closed 2016-02-04 by fixingthingsguy

marcelstoer, 2016-02-10:
Since this question comes up quite frequently do we need to extend `adc.md`?

TerryE, 2016-02-10:
Maybe @fixingthingsguy can propose a PR including this?

fixingthingsguy, 2016-02-11:
Hello TerryE and Marcel:
I can help with some comments but I'm not a reliable coder, haven't been doing that for some time and am now a "user" of the fine stuff this community is creating. I can certainly help with independently testing as   needed.
 jmatson has taken out a placeholder for these kinds of issues that has come to his attention related to adc(tout) and perhaps other similar items.
My thought is that there have been references by others including TerryE for creating a configuration for example for say a 512K ESP01. I am wondering if there is a similar argument to be made for making choices in the esp_init_data for specific uses. Building for each specific need such as the ADC(Tout) is not practical (ie, frightanic builds for ex) and will lead to a proliferation of "images" and make it difficult for development, debugging or testing. 
Perhaps, there is a place for a general purpose configuration  tool for use to choose parameters in the esp_init_data.   Use cases could be added as needed when the parameters in the esp_init_data are mined by  users  for new feature activation. It will also keep the SDK/Nodemcu always consistent, ie. the features turned on in the dev load and master load are those enabled in the esp_init_data. 
Hope that helps.

jmattsson, 2016-02-11:
I have been contemplating adding an `adc.rewrite_byte107(adc.TOUT | adc.BATT | val)` command to allow changing the behaviour post-flash. It would be a once-off sort of deal and it would include a chip-reset to make the change actually take effect.

I'm not entirely comfortable with this approach, but it might be the lesser evil / most user friendly way to deal with the ADC situation.

And before anyone gets their hopes up, chances are I won't have time to look at actually implementing this for a few weeks as I'm rather busy with $work. Assuming people actually think it'd be a good thing in the first place.