nodemcu/nodemcu-firmware #1012: Nodemcu 1.5.1 truncated uart receive

YannickRiouRobotics, 2016-02-04:
Hello, 

I am having a problem with the recent version of nodemcu 1.5.1. everything is working fine except the uart driver. It seems that when I send data to the module via uart, it gets truncated. As an example, if I send this : **c7012345678910112233445566778899** on uart, and I print it on the esp8266, it show up like this : **5678910112233445566778899c701234**.

For information, I am using uart at 921600, 8, N, 1.

The uart was working great on this release : 0.9.6-dev_20150704

I know that it is normal that there is bug because I am using a dev version not yet released, but I wanted to point out this problem.

Regards, 

Yannick

Hellsy22, 2016-02-04:
Why you use ```uart.on("data", 0...``` instead of ```uart.on("data", "\n",``` or any other kind of stop symbol? Without it data will be chunked - actualy with ```uart.on("data", 0...``` callback can be invoked for every symbol recieved.

devsaurus, 2016-02-04:
@Hellsy22 nailed it down.
Previous firmware had different timing characteristics, thus you didn't observe such truncation there. It's just a matter of probability.

YannickRiouRobotics, 2016-02-05:
Thanks guys, it is working great ! I was using `uart.on("data", 0...` because the data that I could receive can contain '\n' in useful data. But now it is working well ! 
Thanks also for the explanation, I was kinda lost about the reason why this is happening !

Great work by the way for the new features (enduser, http, etc).

closed 2016-02-05 by YannickRiouRobotics