nodemcu/nodemcu-firmware #1015: New module for easy bit banging?

romanchyla, 2016-02-06:
Following the contributor's guide, I'd like to ask if there is an interest in having a generic module for reading GPIO sensors. I've built it for reading hx711 (which is already inside nodemcu), but then I made it more generic - so it might be more useful for any sensor where the data has to be read using 'bit banging'

https://github.com/romanchyla/nodemcu-firmware/blob/hx711/app/modules/easygpio.c

romanchyla, 2016-02-07:
alright, I took the chance - please see #1017

marcelstoer, 2016-02-07:
Thanks for contributing.

> alright, I took the chance - please see #1017

And where are we supposed to leave comments now, here or at the PR? I suggest you close either of the two.

closed 2016-02-07 by romanchyla