nodemcu/nodemcu-firmware #1019: hello !  ws2812B  Soft AP not works

Retroaktiv, 2016-02-07:
hello!

I have problems with the DEV version of nodemcu. 
I used DEv with:
adc,bit,bmp085,cjson,coap,crypto,dht,file,gpio,hx711,i2c,mqtt,net,node,ow,pwm,rtcfifo,rtcmem,rtctime,sntp,spi,tmr,tsl2561,u8g,uart,ucg,wifi,ws2801,ws2812
Modules.
The Softap starts and the Index.html works. 
On the ESPlorer  see the commds from my handy ( ~00ff2a)
but the Led Stripe is not lit.

With nodemcu_integer_0.9.5_20150318.bin works my Lua-Skript ind the Stripe lights.

I I tried several and some modules, but no change. Where can my fault?

I hope I can write here!
Thanks for your help. Sorry for my bad English
Retroaktiv

devsaurus, 2016-02-07:
The WS2812 module is currently only able to drive LEDs via GPIO2. See #966 for details.

Retroaktiv, 2016-02-07:
OMG ! 
Thank you !

closed 2016-02-08 by TerryE