nodemcu/nodemcu-firmware #1020: i2c.write sending only first character of String

mamert, 2016-02-07:
Please tell me if I'm simply doing something wrong.

The following works fine:
```lua
	local duino = 0x40
	i2c.start(0)
	i2c.address(0, duino ,i2c.TRANSMITTER)
	data:gsub(".", function(c) -- sending 1 character at a time
		i2c.write(0,c)
	end)
	i2c.stop(0)
```

but just doing `i2c.write(data)` instead of the gsub does not, and only the first character is received.

Te ESP is running a [custom](http://nodemcu-build.com/) NodeMCU build based on master commit c8037568571edb5c568c2f8231e4f8ce0683b88
The receiving device is an Arduino slave running a `while(Wire.available()) Wire.read()` loop registered in `Wire.onReceive`.

devsaurus, 2016-02-07:
> but just doing `i2c.write(data)` instead of the gsub does not, and only the first character is received.

You're executing `i2c.write(0, data)` in this case, right?
What is returned by this call? The expected number of sent bytes or `1`?

mamert, 2016-02-08:
Yes, my mistake. Corrected the post. 
Anyway, `i2c.write(0,data)` returns 0. I recall that I could once get it to send 2 characters (can't reproduce it atm), and the write call consistently returned 1.

Further testing reveals when using the loop
```lua
	data = "SetTime:1453962829"
	data:gsub(".", function(c) -- sending 1 character at a time
		print("Sent "..i2c.write(0,c))
	end)
```
the output is:
```
Sent 0
Sent 0
Sent 1
Sent 1
Sent 0
Sent 0
Sent 0
Sent 1
Sent 0
Sent 1
Sent 0
Sent 0
Sent 0
Sent 1
Sent 1
Sent 1
Sent 1
Sent 0
```
For the string `"QwErTyUiOpAsDfGhJk"` (and `"qWeRtYuIoPaSdFgHjK"`, too) I get:
```
Sent 0
Sent 0
Sent 0
Sent 1
Sent 1
Sent 0
Sent 0
Sent 0
Sent 0
Sent 1
Sent 0
Sent 0
Sent 1
Sent 1
Sent 0
Sent 1
Sent 1
Sent 0
```
Every time, also across restarts. All characters are received on the other end.
Further tests pending your advice.

devsaurus, 2016-02-08:
This looks odd. I'll try to reproduce this here.

mamert, 2016-02-08:
Exact same behavior (same "bytes sent" sequences) on another board with different custom build based on same commit number.

TerryE, 2016-02-08:
@mamert, I am going to be looking at i2c for other work, so I will think about this. At you sure that this isn't a timing issue in the driver on the Arduino end? The I2C protocol is master clocked so bytes are pumped out as 8 bit sequences with no delays between bytes or stop bits out the like. Is this what the recurved is expecting?

TerryE, 2016-02-08:
Yup the I2C code sends multiple bytes so long as the slave replies with an ACK. (See `app/modules/i2c.c:L110`, `app/platform/platform.c:L482`, `app/driver/2c_master.c:L204`.   If it doesn't then only the first byte is sent.  

It looks to me as if your Arduino driver is not sending the ACKs for each byte received.  If we'd had a multi-byte send issue, then this issue would have cropped up long before now.

mamert, 2016-02-08:
@all Sincere apologies, my mistake. I have been using incorrectly placed resistors. The succesful single byte transmissions have thrown me off. I'd be going red with shame over such a noob mistake if I weren't, in fact, a noob.
So: problem doesn't exist.

closed 2016-02-08 by mamert

TerryE, 2016-02-08:
Don't worry.  We try to avoid providing the sort of help that StackOverflow and esp8266.com provides for application developers, but this could well have been a firmware bug and hence in our bailiwick.