nodemcu/nodemcu-firmware #1024: enduser_setup does not seem to honor failed wifi authentication after one successful WiFi connection

vishwaraj00, 2016-02-08:
I have connected NodeMcu to WiFi by using enduser_setup module accessing the softAP in my Android device and entering SSID and password in the html setup page. I faced two situations afterwards:
1: If there are a certain number of authentication failures, NodeMCU keeps retrying and does not connect to WiFi successfully.
2: If the WiFi itself is out of reach or the device itself moves to a different WiFi zone, NodeMCU keeps trying and cannot find a valid WiFi SSID.

In both these cases, NodeMCU does not fall back to the original WiFi setup state. Hence, my device remains out of reach and the only solution seems like I need to re-flash, so that I can access the SoftAP and the html setup page.

marcelstoer, 2016-02-09:
I assume you're familiar with http://nodemcu.readthedocs.org/en/dev/en/modules/enduser-setup/ (since you use the sample code). Then I guess you also know you can switch the module back to 'manual' or stop it altogether?
Do you want to know how to apply those functions or do you suggest there's a bug in @robertfoss' implementation i.e. a bug in `enduser_setup.c`?

robertfoss, 2016-02-09:
Currently the enduser_setup module takes no responsibility for saving or going back to any original settings after it has been launched.

The module will keep running until a successful connection has been made, so if you visit the captive portal you should be able to submit any WiFi connection details you want.

Does that make sense? Or am I misunderstanding the issue?

fixingthingsguy, 2016-02-09:
([vishwaraj00](url))   I've been playing around with this feature, feature is working as advertised. Here is my experience. Tried with several hw. variations including the ESP-01 and nodemcu branded hardware.
**Log in sequence:**
-Powered up(device) in default mode, captive portal presented(on smartphone), logged in success, using my home wifi.
-Powered down, went across to city library, powered up device, captive portal presented, logged in success, even accepted null password at library
-Returned home, same sequence.
 

**Wifi failure mode;**
-While operating at home, powered down home wifi. (I don't remember now if it presented portal,should have)
-Turned back up home wifi, device is connected using last wifi ssid/pwd.
**Other**
Enduser_setup code used as in your write up([vishwaraj00](url))
Load used was frightanic build off dev load 11/22/2015 with enduser_setup module selected.

robertfoss, 2016-02-09:
@fixingthingsguy  So the failure is that it it connects to your home wifi after the home wifi powers up again?
I don't think I understand.

marcelstoer, 2016-02-09:
My main take away from @fixingthingsguy's comment was 

> feature is working as advertised

:smile:

fixingthingsguy, 2016-02-10:
robertfoss  -Rest assured the feature is working as documented.
Regarding the below test, the initial condition was that the ssid/pwd had been already set up for the home and then shutting down and restarting the home wifi router resulted in a success reconnection by the device that was using the enduser_setup feature.
**Wifi failure mode**
•While operating at home, powered down home wifi. (I don't remember now if it presented portal,should have)
•Turned back up home wifi, device is connected using last wifi ssid/pwd.

I don't claim to understand vishwaraj00 s problem description. He needs to provide some more detail I would think with initial conditions.

vishwaraj00, 2016-02-11:
I think I need to give in more details after reading some of your comments/advice: "Feature is working as advertised" and "Need to provide some more detail", hence following up on that. I hope I am able to explain it this time.

@marcelstoer , You are spot on, I am expecting more documentation or help on how to use enduser_setup in manual mode as the current documentation is very limited and not descriptive :-1:  
[enduser_setup.manual()](https://nodemcu.readthedocs.org/en/dev/en/modules/enduser-setup/#enduser_setupmanual).

**ISSUE:**
I have experienced the behavior as explained by @fixingthingsguy except that in my WiFi failure scenario, the old WiFi SSID is permanently switched off and will never be back up again. So, the user has to connect to some other SSID WiFi through the SoftAP and captive portal. Now, once a WiFi failure occurs, I do not see the softAP at all in my smartphone. So, I am stuck with a situation where my old WiFi SSID is down and I cannot enter the credentials for a new WiFi SSID.

**FLOW OF ACTIONS:**
1: I saved the WiFi credentials to the SoftAP and observe that a valid ip address is allocated to NodeMCU. Works great.
2: I transfer my device to an area with new WiFi ssid. I know that I need to reconnect to the SoftAP of NodeMCU and re-enter the credentials. But, I cannot see the softAP again.** [<-- Issue is here]**
3: I move back to the old working area and as @fixingthingsguy  said, it works back again, but moving back to the old WiFi SSID is not an option for me.

fixingthingsguy, 2016-02-11:
@vishwaraj00 
I did exactly the same as you described in Flow of Actions. Although you didn't state it I'm guessing you also powered it down and then re-powered at the new location. I mimicked all this just a short while ago using a mobile hotstpot on my smartphone as the wifi source. And it all works and I secure an IP address both times. I have done this on two different nodemcu hw devices. 
So I'm not sure what is going on for your test. I'll dig around some more; @robertfoss is the designer of this feature, maybe he can suggest other tests to gather data.

vishwaraj00, 2016-02-11:
@fixingthingsguy, you are getting it wrong. 
I took NodeMcu to another location and then, I have to connect to **another** WiFi not the old one. For that the SoftAP is not coming up so that I can put my credentials of the **another** WiFi.

robertfoss, 2016-02-11:
But did you restart the NodeMCU when you moved it?

The enduser_setup code has to be run every time you want the NodeMCU to
connect to a new WiFi.

vishwaraj00, 2016-02-11:
@robertfoss , After the **old** WiFi goes down, NodeMcu goes into an endless loop and tries to connect to the same Old WiFi again & again, even after I restart the NodeMcu. I believe that the expected behavior should be that NodeMCU shows the softAP again and let me enter the credentials through the captive portal for the **new** WiFi. So, it looks like a behavioral issue to me.

But anyways, it would be acceptable for me to use the manual mode instead as @marcelstoer  suggested. Could you folks please suggest me some working code/examples like how I use the manual mode? As I said in my previous comment, I feel the documentation for the manual mode is quite insufficient for a general understanding.

robertfoss, 2016-02-11:
But are you executing enduser_setup.start() after the NodeMCU has been
restarted? Maybe you could show us the lua code you are running and which
filesnames the lua snippets have.

vishwaraj00, 2016-02-11:
The code I mentioned in the issue is the only code I wrote, saved to enduser_setup.lua, and then simply Run it on NodeMCU. After that the WiFi connection attempts never stops, once I follow the reproducing steps described above.

It's fine, I'll try in manual mode once. But I might need some more examples before I proceed upon that. Can anybody help me?

robertfoss, 2016-02-11:
Alright, there's your problem. Put the same code in init.lua and it will be
run upon every restart of the NodeMCU.Only code that is in init.lua or is
called from init.lua will be executed after a reboot.

Does this solve your issues? If not then what happens when the NodeMCU has
restarted?

vishwaraj00, 2016-02-11:
I didn't know about this init.lua thing. But, if I put the same code in init.lua then won't it ask for WiFi credentials everytime NodeMCU has been restarted?  I wanted the password to be asked only when the credentials that are already entered are not working for connecting to any WiFi successfully.

robertfoss, 2016-02-11:
So NodeMCU will only every try to connect to the last WiFi it was set to
connect to. If you move the NodeMCU to a location where the old wifi is no
longer available and you want it to connect to a new WiFi, you will have to
make sure that enduser_setup.start() is run _somehow_. Either run it
manually, or make sure the code is in init.lua and restart the NodeMCU
board.

vishwaraj00, 2016-02-11:
Okay, I'll try that.

Thanks a lot to all. You can now close this issue.

closed 2016-02-11 by vishwaraj00