nodemcu/nodemcu-firmware #1025: The linker script places libc code into iram rather than irom

pjsg, 2016-02-08:
The current linker script fails (at least in my environment) to put the libc code into irom. Suggest that something like the following snippet be used instead. It also isn't clear to me whether the libgcc stuff could also be moved out. It also isn't clear what the performance tradeoff is.

    /* SDK doesn't use libc, libm functions, and are therefore safe to put in flash */
    */libc.a:(.text .literal)
    */libm.a:(.text .literal)
    /* end libc functions */

I'm building using the docker container: marcelstoer/nodemcu-build  -- which (AFAIK) just unpacks the tools/esp-open-sdk.tar.gz and uses it.

I'm failing to build the floating point version (I have a couple of extra modules with small iram requirements) -- there is a bunch of extra stuff that gets inserted into the iram in that case.

TerryE, 2016-02-08:
@pjsg Phillp, surely we only need iram to be large enough at the moment we have abut 2Kb headroom.  There are performance advantages in having hot code in iram, and given that Lua's default numeric type is double, then there's a good argument for leaving the libc routines that we use in iram.

pjsg, 2016-02-09:
@terryE In the floating point build, we are right on the edge and anything can push it over. I note that the iram consumption of dev is more than in master due to a change in the linker script. I'm not convinced that this was intentional from the associated commit comment:

https://github.com/nodemcu/nodemcu-firmware/commit/d5f71eb0c633a039cdd6b70dc93c44641b5b11ca

The problem is that the new script doesn't actually put all that stuff in irom. There is a missing colon (AFAIK).

TerryE, 2016-02-09:
Yup, the script should read
```
*/libc.a:*.o(.text* .literal*)
```
and on my build this drops the iram .text segment from 0x7d7b to 0x752b -- that's an extra 2,128 head room.  I've also checked the frequency of usage and to be honest all of the common RTL routines are pulled out of ROM anyway, so on reflection I think that you are right.  This is worth doing.  A good catch. Thanks :smile:

pjsg, 2016-02-10:
Further, I think that the chip does have an instruction cache on it -- so the execution speed from flash is the same as it is from ram *provided* that there is minimal cache churn. A simple test shows that a loop runs at the same speed whether or not the code is in flash. 

It isn't clear how big the icache is.... trying to figure it out....

TerryE, 2016-02-10:
> Further, I think that the chip does have an instruction cache on it.

Ditto, and the [Xtensa LX Overview Handbook](http://ip.cadence.com/uploads/pdf/xtensalx_overview_handbook.pdf) describes the various cache options, but like the ARM, the LX is a macro design with lots of configurable options that the chip manufacturer (in the case Espressif) can select, and Espressif keeps this detail closed source so we can only guess.

TerryE, 2016-03-03:
I added this to one of my PRs last week.

closed 2016-03-03 by TerryE