nodemcu/nodemcu-firmware #1027: Even parity not working correct on net.TCP connection

SerdarMSC, 2016-02-09:
Hello. First, sorry for my english :)

my device need 115200E1 baud rate. My code;

`uart.setup(0,115200,8,1,1,0)
sv=net.createServer(net.TCP, 60)
global_c = nil
sv:listen(9999, function(c)
	if global_c~=nil then
		global_c:close()
	end
	global_c=c
	c:on("receive",function(sck,pl)	uart.write(0,pl) end)
end)
uart.on("data",4, function(data)
	if global_c~=nil then
		global_c:send(data)
	end
end, 0)`

normally (ftdi or bluetooth interface) i send com port to device this block (Hex);
`05 24 06 24 0D`
Receive form device to com port;
`05 24 54 48 58 0D`

if i use esp8266, Receive form device to com port;
`05` (only receive 05)

i was change my code;
`
uart.on("data",4, function(data)
	if global_c~=nil then
		global_c:send(data)
	end
end, **1**
`
and i receive ;
`05 24 54 48 58`

Even parity bit `0D` not receiving.

if i use net.UDP;
line 1 `05 24 54 48 58`
line 2 `0D` 

my device working but its UDP, i need TCP. 

Why i recieve max lenght 5?

Regards,

Hellsy22, 2016-02-09:
Again the ```send``` problem. You try to send more data before ```sent``` event. UDP just not waiting for confirmation.

You can modify lua code to accumulate data till ```sent``` event and then firing it.

dvv, 2016-02-09:
I believe we should `nil` the `send` method after the first invocation! )

marcelstoer, 2016-02-09:
http://nodemcu.readthedocs.org/en/dev/en/modules/net/#note

closed 2016-02-09 by marcelstoer

TerryE, 2016-02-09:
@SerdarMSC, Serdar you really need to direct this sort of question to a support forum, however, I have tidied your post so that others can read it more easily.