nodemcu/nodemcu-firmware #1031: SPIFFS image flashing

DesperateProgrammer, 2016-02-10:
Hello,

I am currently trying to generate a SPIFFS image to flash with the nodemcu-firmware using the spiffy ( https://github.com/xlfe/spiffy ) tool with spiffs 0.3.2 ( https://github.com/pellepl/spiffs/releases/tag/0.3.2 ) as this seems to be the version to be used by nodemcu.
This generates an image, that can only errornous read by nodemcu. Meaning: all filenames begin with \001 and a wrong filesize. (usually way too big)

I then moved to another spiffy branch within https://github.com/CHERTS/esp8266-devkit/tree/master/Espressif/examples/ESP8266/nodemcu-spiffy 
with the same problem.

To check the alignment (because I am not sure the error is named "me" ;-) ) 
The firmware 0x10000.bin file is 0x71318 Bytes long after an clean build. Meaning that the spiffs should be located at 0x84000 = ((0x71318 +0x3FFF) & 0xFFFFC000

Is the spiffs version 0.3.2 correct for nodemcu?
Is the alignment correct?
Is there any modification I did not catch against 0.3.2 spiffs?
How is your opinion about fs generating in the Makefile, should this be added? I would implement a patch allowing to build the fs (either as option or default) once i figure out whats failing right now

Greets

DesperateProgrammer, 2016-02-10:
Caught at least one difference to spiffs 0.3.2:
spiffs_nucleus.h in nodemcu does not contain the same manual alignment then 0.3.2

https://github.com/nodemcu/nodemcu-firmware/blob/dev/app/spiffs/spiffs_nucleus.h#L419
vs.
https://github.com/pellepl/spiffs/blob/0.3.2/src/spiffs_nucleus.h#L422

jfollas, 2016-02-11:
After a tip from @jmattsson, I had success using https://github.com/DiUS/spiffsimg instead of Spiffy.

TerryE, 2016-02-11:
@jfollas, hummn -- you comment is apt. We should include a link to `spiffsimage` somewhere in our build instructions or FAQ. For my to do list.

DesperateProgrammer, 2016-02-11:
@jfollas, thank you for the hint. Indeed, this did work.

@TerryE, i am currently making efforts to include spiffsimg as a submodule and the makefiles to generate the fs automatically, if there are files in a files folder.
I'll commit it when i'm done. 
Currently I am at the point that it is running locally after fixing an issue in spiffsimg when executed in a VM (mmap with MAP_SHARED not supported)

Now i'll make that clean into the right directories 
submodule with /tools Outputting into /bin
Current state of work for this ishttps://github.com/DesperateProgrammer/nodemcu-firmware/tree/master/tools

A question: What would be the preferable folder for the fs template. Right now i have it in /tools/files and every file there gets embedded into the spiffs image. But i don't think that this is the right place.

TerryE, 2016-02-11:
Maybe the other committers have a view on this, but my immediate thought is to have a `local` subfolder and `local/fs` within that.  (I could see us adding in future `local/modules` as well so that the developer puts any custom modules in this and they are automatically added to the build).  Anyway that's my thinking.  Marcel, etc., might have a different view, so please comment.

DesperateProgrammer, 2016-02-11:
Until the decision, i have made all necessary changes in a fork, that can be changed to use the folders decided uppon without much work.

[https://github.com/DesperateProgrammer/nodemcu-firmware/compare/1dfee7452b871a62073b2ea47f95fdd631ecaa7c...9b3d1baa6b91f3e0d99e2e0335bc790ff3c6ee59]

Comments are welcome

DesperateProgrammer, 2016-02-14:
Created a pull request.
Did take a bit longer for cleaning because i am not firm with git yet. (and thus a lot of commits that are just to get back on a clean diff ;-) )

TerryE, 2016-02-14:
Hi.  Since this is your first contribution, you are still on the learning curve.  Let me help you.  I'll take a look tonight and give you some detailed feedback.  But the PR in its current form isn't ready and will need resubmitting.  Drop me a PM on esp8266.com and we can hook up by email / skype (if you are comfortable with spoken English or IRC -- my German is non-existent, sorry.

marcelstoer, 2016-02-15:
> i am not firm with git yet

Welcome to the club :wink: That's why we created https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md#working-with-git-and-github

boids, 2016-02-17:
@DesperateProgrammer, I used your branch created 0x78000.bin(lua bin) and I'm using following esptool command load the bin files in
"./nodemcu-firmware/tools/esptool.py --port /dev/cu.wchusbserial14210 write_flash 0x00000 0x00000.bin 0x10000 0x10000.bin 0x78000 0x78000.bin"
but when I run the board, it still return "NodeMCU 1.4.0 build 20151006 powered by Lua 5.1.4 on SDK 1.4.0
lua: cannot open init.lua"

what went wrong, please help!!!!

jfollas, 2016-02-17:
@boids If you try flashing just the 0x78000.bin again, does it work?

When I played around with "triple bin flashing" a while ago, I found that the NodeMCU firmware would not recognize the filesystem and format upon first boot (which prompted be to write https://github.com/nodemcu/nodemcu-firmware/issues/995). Wondering if this is what's happening in your case, too.

gsker, 2016-02-17:
I'm working with this too and haven't been very successful.   Even flashing a 0x780000.bin that should look like an spffs file system image.  I doubt it is a good image given some other digging.
    https://github.com/DesperateProgrammer/nodemcu-firmware/pull/1#issuecomment-184567414

boids, 2016-02-17:
@DesperateProgrammer, flash 0x78000.bin alone works! thank you!!!

jfollas, 2016-02-17:
This is what I was afraid of.

With #FLASH_SAFE_API defined, the block starting at line 126 of https://github.com/nodemcu/nodemcu-firmware/blob/dev/app/user/user_main.c gets executed when `flash_safe_get_size_byte() != flash_rom_get_size_byte()`, which formats the filesystem. So, in this case (and in the case of #995), flashing the SPIFFS image at the same time as the other two .bin files appears to result in the SPIFFS image getting overwritten by the start-up format.

jmattsson, 2016-02-17:
If you don't lie about the flash size when you do the first write, you won't run into that problem ;)

E.g. for an ESP-12(E):
`esptool.py --port /your/portHere write_flash `**`-fs 32m`**` 0 bin/0x00000.bin ...`

DesperateProgrammer, 2016-02-17:
That is indeed nice to know.
Right now I have a little problem getting the time at work for this, but it might be a reason for another problem i have had with flashing from another uC. Need to check this once I am done keeping the existing things up and running ... and reading more about git :-D

gsker, 2016-02-18:
Sorry for the typo in my last comment -- 0x78000 not 0x780000.

@jmattsson  Who lies about the flash size?  I've seen that syntax around before and wondered why one  would say they had 32m of flash when they only had 4?

I am not doing any such lying, but if I flash 3 files in a row then the format happens when it starts up. 
If I then flash the 3rd file after the format is done, then that appears to work.
The three-file write:
   esptool.py -b 345600 write_flash 0x00000 0x00000.bin 0x10000 0x10000.bin 0x78000 luafiles.img

Then later with the single:
    esptool.py -b 345600 write_flash 0x78000 luafiles.img

-rw-rw-r-- 1 gsker gsker  33904 Feb 17 18:29 0x00000.bin
-rw-rw---- 1 gsker gsker 416768 Feb 17 18:29 0x10000.bin
-rw-rw---- 1 gsker gsker 196608 Feb 17 18:55 luafiles.img

Where does the image have to appear to prevent the auto-format?
I'm sure I'm doing something simple wrong.

My location calculation:
~> echo "scale=0; obase=16; ( $rom + 81919 )/16384 * 16384"| bc -q
scale=0; obase=16; ( 416768 + 81919 )/16384 * 16384
78000

jmattsson, 2016-02-18:
32m = 32mbit = 4mbyte. Flash sizes are typically expressed in kilo/mega-bits rather than kilo/mega-bytes for (?) historical reasons, and that can cause no end of confusion...

On the ESP8266 the header of the flashed firmware contains information about the flash size, mode and speed of the flash chip. The ROM bootloader relies on this information when it boots the firmware. If this information is incorrect, various Bad Things(tm) can happen (including a complete failure to boot, crashing when attempting to access flash which doesn't exist, or having the SDK clobber its configuration data into the middle of the filesystem). NodeMCU, when booting, inspects the size parameter stored in that header and compares it to the reality of the flash on the chip, and in the case of mismatch *updates the information in the header*. And if there was a mismatch, NodeMCU tries to play it safe and reformats the file system to get the flash into a known state, in case the SDK had corrupted it.

When flashing with esptool.py (and the other ESP flash tools) you can (and should) explicitly declare the flash parameters of the chip you're installing the firmware to. Doing this means that the bootloader can reliably run the firmware you flash, the SDK can reliably find (and store) its configuration data in the right spot, and NodeMCU doesn't see a need to reformat the filesystem.

jmattsson, 2016-02-18:
You might also find #1063 useful.

gsker, 2016-02-18:
Excellent info. Thank you! 

That's great information and sure enough, adding -fs=32m (which I would have sworn I tried before) worked!  

Now I'm gonna have to stretch my git-fu and  figure out how to include that pull request into my source tree. :-)

karrots, 2016-02-18:
Slightly advanced way but it works pretty well.

https://help.github.com/articles/checking-out-pull-requests-locally/

gsker, 2016-02-18:
Thank you!
That was the first return I got in a search for that functionality.
But it isn't quite what I want to do.
It's not a pull request on my repo; It's a pull request on the nodemcu repo. 
So it doesn't show up in my repo as a pull request and none of that seems to apply. 
I'm sure I've got it all horrible confused in my head... I'm trying.

I have a cloned repository on my desktop that I build from.  I would be asking, "how I can take that one commit and merge it ijn a branch in local repository".

With that said, it's way too many steps and it might work well, but it's not scrutable at this point in my learning. :-(   

It's about 6 lines of code. I put it in manually.    :-)

> =file.fscfg()
491520  3686400

obase=16
491520
78000

Cool!

karrots, 2016-02-18:
You just need to add the main nodemcu repo as a second remote and then it
will work.

gsker, 2016-02-18:
:-)  Nope. Too many hard words.  Thanks again though. I **really** appreciate it and I know it'll come clear eventually.

marcelstoer, 2016-02-18:
We documented all of that for you, and others, in https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md.

DesperateProgrammer, 2016-02-18:
I hope this time the PR does match the requirements.

I just now have seen, that the Docker image does not know the bc command.
[http://linux.die.net/man/1/bc]

Adding the bc package would be required there to name the spiffs image correctly (does the dezimal to hex conversion within the makefile shell.

marcelstoer, 2016-02-18:
> Docker image does not know the bc command

Now it does: https://github.com/marcelstoer/docker-nodemcu-build/blob/master/Dockerfile#L11 Docker Hub is currently building the new image.

pjsg, 2016-02-18:
@marcelstoer Can you also combine the spiffs image into the single .bin file that you create in the docker image?

marcelstoer, 2016-02-18:
@pjsg pls create an issue at https://github.com/marcelstoer/docker-nodemcu-build for this.

TerryE, 2016-02-19:
> the Docker image does not know the bc command

But it does know bash, perl and python

gsker, 2016-02-19:
And esptool.py can probably deal with deal with decimal so your prior $(()) format should work fine. The only thing that the bc was used for was the conversion to hex if I'm remembering right.

vsky279, 2016-02-19:
Interesting issue: I was not able to make the spiffs image work after being flashed using "ESP8266 Flasher".
Flashing spiffs image with `esptool.py` works like a charm.
Very usefull as a workaround for the issue #898.

I enjoy `=file.fscfg()`. Good idea, thx.

gsker, 2016-02-20:
Thanks all for the git encouragement.  I did figure how to get pulls into my local.  Not really understanding it but getting better.

DesperateProgrammer, 2016-02-20:
I am going to remove the dependency to bc and using bash and printf(1) instead

Flashing the file also on make flash is pretty simple to add. 
I think for this to be really usefull and not cause too much confusion we will need:
* the spiffs image only to be created if local/fs is populated
* the flshtool to auto-detect the esp's connected flash size (so that after using make flash it is not just formated again)

Otherwise I would leave flashing the spiffs image manually.

What's your all opinion about this?

TerryE, 2016-02-20:
IMO, I would avoid trying to flash as part of the build. We need a separate target, say make flash or make spiffs, that is easily memorable.

jfollas, 2016-02-20:
I read this as whether `make flash` should automatically include the SPIFFS image if it's available (i.e., so you'd flash 0x00000.bin, 0x10000.bin, and, say, luafiles.img to 0x7c000 all at the same time). 

If that's possible in a `make` target (dynamically determining whether to include the SPIFFS image in the esptool arguments, and where to flash it to in the address space), then I certainly see the convenience. It would only be performed if someone took advantage of the local/fs feature.

marcelstoer, 2016-02-20:
In #595 the current collaborators more or less agreed to 
- remove the `flash` target
- replace the current esptool.py copy we host in our repository with a trimmed down version from @jmattsson which does only build the image (i.e. has no flash capability)

I, or someone else, still need to actually implement that but I think it's helpful to keep this in mind when talking about making changes to `make` targets and (NodeMCU) esptool.py.

TerryE, 2016-02-20:
Sorry, my bad, I was using `flash` in the wrong context: flash filesystem rather han "flash to device".  I agree with #595.  But a spiffs (this is less ambiguous) target does make some sense.

DesperateProgrammer, 2016-02-22:
#595 is a game changer for the auto generation.
I agree, that it does not make sense to add it into the flashing target in this context.
And also i don't think adding a new bin file would be the way to go then.

I think the best approach is to append the spiffs image padded to the 0x10000 image in the first step either by make target, or by existance oif local/fs files.
The later one would have the charme, that it does not change anything for existing workflows, and yet is easy to use.

Both ways would not interfere with any replacement for esptool.py or with the goal to combine both existing .bin files into one firmware image. 
For the same reason it would not require any change to any web how-to about flashing right now, be it with the espressif flasher or esptool.py as it is now.

TerryE, 2016-02-22:
@DesperateProgrammer, the reason for #595 was to bring the esptool that we depend on for our building under our change control.  

I personally think that adding a new bin file is the correct way to go because a production user might want to update the spiffs image on a more frequent basis than the 0x00000 or 0x10000 images, so there is a good argument for keeping this imaging process separate.

NicolSpies, 2016-02-22:
@DesperateProgrammer , I agree with TerryE for keeping it separate as I update the spiffs image regularly. I am not a C programmer and only use NodeMCU build using frightanic for custom builds having the effect that I will always build the spiffs image alone.

karrots, 2016-06-07:
Possibly fixed by #1226

closed 2016-11-19 by marcelstoer