nodemcu/nodemcu-firmware #1033: 1-wire issue with 0.9.6-dev_20150625

pilhuhn, 2016-02-10:
I am using an Adafruit Huzzah and reading data from ds18b20 - so far so good.
Due to my stupidity I had to reflash the software and decided to use version 0.9.6-dev_20150625   to find out that the example from http://nodemcu.readthedocs.org/en/dev/en/modules/ow/#example no longer works. Going back to 0.9.5_20150318 fixes the issue for me. I did not yet try a newer version though.

No working is that the data read here (from the example)
     ```
 data = string.char(ow.read(pin))
          for i = 1, 8 do
            data = data .. string.char(ow.read(pin))
          end
```
has all 8 bytes set to 255.

TerryE, 2016-02-10:
Sorry but we are now up to SDK V1.4 on master and SDK 1.5.1 on dev.  Please see #719 for an explanation of how to seek programmer support on old versions.

closed 2016-02-10 by TerryE

pilhuhn, 2016-02-11:
While I understand that the report is not against HEAD, it was not a 
question about programmer support.
I have running code against 095, that breaks afterwards with a later 
nodemcu version.
If this is a known ow-Issue and fixed already then fine.