nodemcu/nodemcu-firmware #1034: wifi.setmode(wifi.NULLMODE) reports "wrong arg type"

joysfera, 2016-02-10:
The subject says it all. This happens with the most fresh build of dev tree.
"NULLMODE" is visible in the blob so it is defined yet it's not accepted.

dnc40085, 2016-02-11:
I also noticed that `wifi.setmode()` didn't accept `wifi.NULLMODE` as an argument when I was writing PR #1018. So I [added a fix](https://github.com/nodemcu/nodemcu-firmware/pull/1018/files#diff-708acae44868e2f538336aac253c2bc5R219) for it.

marcelstoer, 2016-02-23:
Thanks @dnc40085, your PR will fix this.

closed 2016-02-23 by marcelstoer

devyte, 2016-03-05:
Hi, I'm a bit confused with this. Was the fix mentioned here put in, is it still pending, or was it lost?
NodeMCU custom build by frightanic.com
	branch: dev
	commit: 568439291152f81ace202a60d4371070c23bc4a3
	SSL: true
	modules: adc,bit,cjson,crypto,file,gpio,net,node,pwm,tmr,uart,wifi
 build 	built on: 2016-03-03 15:21
 powered by Lua 5.1.4 on SDK 1.5.1(e67da894)
wifi mode set to wifi.SOFTAP
> =wifi.setmode(wifi.NULLMODE)
stdin:1: wrong arg type
stack traceback:
	[C]: in function 'setmode'
	stdin:1: in main chunk
>

dnc40085, 2016-03-05:
pending

AliKamil, 2016-03-19:
Also has same issue with build from dev. wifi.NULLMODE constant presents, but is not accepted by wifi.setmode:

```lua
NodeMCU custom build by frightanic.com
	branch: dev
	commit: 3d508591e01786ffe002e2aea38e1df17f56c094
	SSL: false
	modules: bit,cjson,file,gpio,net,node,pwm,rtctime,tmr,uart,wifi
 build 	built on: 2016-03-19 10:27
 powered by Lua 5.1.4 on SDK 1.5.1(e67da894)
lua: cannot open init.lua
> print(wifi.NULLMODE)
0
> wifi.setmode(wifi.NULLMODE)
stdin:1: wrong arg type
stack traceback:
	[C]: in function 'setmode'
	stdin:1: in main chunk
> wifi.setmode(0)
stdin:1: wrong arg type
stack traceback:
	[C]: in function 'setmode'
	stdin:1: in main chunk
```

dnc40085, 2016-03-20:
the fix for this is still waiting to be merged

gutierrezps, 2016-04-05:
Same issue here.

marcelstoer, 2016-04-05:
@gutierrezps are you on `dev` branch?

gutierrezps, 2016-04-05:
My built was from 2016-03-28. I've got a build right now, and it's working. Sorry.