nodemcu/nodemcu-firmware #1035: wifi.sta.getap() does not fill the RSSI

joysfera, 2016-02-10:
I normally use the latest official build ("nodemcu_integer_0.9.6-dev_20150704.bin") where the wifi.sta.getap() finds all APs and fills the ssid, rssi, authmode and channel return values.
However, with the latest build of the "master" tree the RSSI is zero.
For comparison, the latest build of the "dev" tree returns correct RSSI.

TerryE, 2016-02-10:
> I normally use the latest official build ... 0.9.6-dev_20150704.bin

The was the last pre-built version, but it is quite wrong to describe this as the _latest official build_ as completely configured builds no longer fit in the 512Kb ESP modules.  We support custom builds as per our [documented processes](http://nodemcu.readthedocs.org/en/dev/en/build/).

closed 2016-02-10 by TerryE

joysfera, 2016-02-10:
Terry, I am afraid you misread what I wrote. Let me try to explain it again: RSSI is not filled in the custom build of "master" tree. It is, however, correct in the custom build of "dev" tree.

TerryE, 2016-02-10:
Yup, but we had a long discussion on how to streamline our release practice.  What we do is to snap a stable dev version into master on a routine basis.  If master has this bug, and dev has already fixed it then we don't do interim cherry-picks from dev into master.  Users can only chose between dev and master, or do a local clone and do the cherry-pick themselves. Sorry, but given that noone funds our work, and developers can download the latest firmware free, we need to prioritise our resources here :smile:

joysfera, 2016-02-10:
I am perfectly OK with running on dev version only (or until it's fixed in master, if running on master is generally preferred). I didn't know about the dev->master snapping, thanks for explaining that.