nodemcu/nodemcu-firmware #1036: u8g library does not include all drivers

joysfera, 2016-02-10:
Documentation says that u8g library contains 20+ drivers (please see http://nodemcu.readthedocs.org/en/dev/en/modules/u8g/#spi-display-drivers ) yet the custom build of "dev" tree that includes U8G contains single driver only: "ssd1306_128x64_hw_spi".
Would it be possible, please, to include the other drivers as well? Or is it possible to select the driver at build process somehow? I use the http://nodemcu-build.com/ service and happen to have sh1106 display.

devsaurus, 2016-02-10:
Your observation is correct - the nodemcu-build service doesn't allow to select specific display drivers at the moment. I don't know if @marcelstoer plans to add this in the future.
Thus you need to enable specific display drivers manually in `app/include/u8g_config.h` like explained [here](https://nodemcu.readthedocs.org/en/dev/en/modules/u8g/#displays) and compile the firmware yourself.

joysfera, 2016-02-10:
Thanks, I'll give the Docker image a try then.

marcelstoer, 2016-02-11:
> I don't know if @marcelstoer plans to add this in the future.

I do.

joysfera, 2016-02-11:
Marcel, that would be just great. Please simply copy the "U8G fonts" input field and name it "U8G drivers" , add a "ssd1306_128x64_hw_spi" default and that's it.

devsaurus, 2016-02-12:
BTW, shall we track feature requests for the build service in this issue list?

marcelstoer, 2016-02-12:
No.

closed 2016-02-12 by marcelstoer