nodemcu/nodemcu-firmware #1038: Why string length of crypto.decrypt is not equal to original length?

smileshop, 2016-02-11:
NodeMCU custom build by frightanic.com
	branch: dev
	commit: 2c687313decd92b8f4bd25961ac7793692df3986
	SSL: false
	modules: bit,crypto,enduser_setup,file,gpio,http,i2c,mqtt,net,node,tmr,u8g,uart,wifi
 build 	built on: 2016-02-10 15:35
 powered by Lua 5.1.4 on SDK 1.5.1(e67da894)

local key = "1234567890abcdef"

cipher = crypto.encrypt("AES-ECB", key, "Hi, I'm secret!")

decypher = crypto.decrypt("AES-ECB", key, cipher) 

print('Original length = '..string.len("Hi, I'm secret!")..' , decypher lengh = '..string.len(decypher))

Result:  Original length = 15 , decypher length = 16

ABonner, 2016-02-11:
The decrypted string appears to have a null character appended to it. Slightly updated testcase below that prints out all the characters in the string.

    plaintext = "Hi, I'm secret!"
    key = "1234567890abcdef"
    cipher = crypto.encrypt("AES-ECB", key, plaintext)
    decipher = crypto.decrypt("AES-ECB", key, cipher)
    print('Original length = '..string.len("Hi, I'm secret!")..' , decipher length = '..string.len(decipher))
    t={}; plaintext:gsub(".", function (c) t[#t+1]=string.byte(c) end); print ("Plaintext: "..table.concat(t, ","))
    t={}; decipher:gsub(".", function (c) t[#t+1]=string.byte(c) end); print ("Decipher: "..table.concat(t, ","))

Result:

> Original length = 15 , decipher length = 16
> Plaintext: 72,105,44,32,73,39,109,32,115,101,99,114,101,116,33
> Decipher: 72,105,44,32,73,39,109,32,115,101,99,114,101,116,33,0

I can take a look later today since I've been digging around in the crypto module already.

TerryE, 2016-02-12:
Hummmn, sounds like documenting a bug is turning it into a "feature" to me :laughing:

jmattsson, 2016-02-12:
When encrypting an arbitrary-length string with a block cipher there is nothing which can remember the original string length without imposing a particular format (e.g. `struct foo { uint16_t len; char str[]; }`). Doing so at the level of the `crypto` module would effectively prevent it from being a general-purpose (en|de)cryption and limited to only that particular format. That would render it largely useless as it would only interoperate within NodeMCU, not with external services and protocols making use of said encryption algorithm.

If you would like to improve the existing documentation to clarify this behaviour even further, we are happy to receive PRs :)

closed 2016-02-12 by jmattsson

TerryE, 2016-02-12:
True.

veljani, 2016-04-20:
I tried everything and you are my last hope! Please help :)

I built from nodemcu-build.com tried with minimal number of libraries and offcourse crypto.

When I try example I get this error:
print(crypto.encrypt("AES-EBC", "0123456789abcdef", "some clear text data",'0000000000000000'))
stdin:1: attempt to call field 'encrypt' (a nil value)

I tried integer and float, ssl included also, nothing works.
I suppose it is stupid mistake and easy solution... but I'm not a smart man :)
Used chip: esp-12e, esp-01.

Thank you in advance. - See more at: http://www.esp8266.com/viewtopic.php?f=6&t=9615#sthash.XRxzz3YG.dpuf