nodemcu/nodemcu-firmware #1039: rtctime.dsleep doesn't set correct bootreason

viperidae, 2016-02-11:
When coming out of a dsleep from rtctime, the extended bootreason is set to hardware reset, not dsleep, like after calling node.dsleep

I saw in the C code that node.dsleep calls a system function, while rtctime.dsleep sets registers manually. Is this just a matter of setting another register?

jmattsson, 2016-02-12:
Thanks for pointing this out. Presumably we would need to squirrel away another magic value into an RTC register before going to sleep. I'll leave a note for my colleague who's the author of this module (he's on leave for another couple of weeks however).

marcelstoer, 2016-04-13:
Johny, is Bernd back yet? I had initially added it to the list in #1146 but I don't think it's critical enough to keep waiting for a fix. Do you agree?

jmattsson, 2016-04-13:
Back and gone again! He's currently in Germany, so you could hound him (@umisef) more easily than I can! :D 

I agree it's not important enough to hold up a release drop.

sfrederiks, 2016-12-14:
A workaround until it is fixed:
rtctime.get() == 0 if it is not wake from rtctime.dsleep_aligned else it gives the time.

closed 2019-06-21 by stale[bot]