nodemcu/nodemcu-firmware #1045: Enable 250kbit baud rate in Uart module

marki555, 2016-02-13:
Why is the 250kbit baud rate not enabled in the Uart module? I see it defined in ` app/include/driver/uart.h ` as `BIT_RATE_256000 = 256000`, but not in `platform_uart_setup()` function in `app/platform/platform.c`.
Is it possible to enable also this bitrate? It is used for example in the DMX512 serial protocol.

dnc40085, 2016-02-14:
Unless I'm mistaken, I believe this same question was asked previously in issue #856

marki555, 2016-02-14:
Yes, seems to be the same... however I don't see any reason why it couldn't be implemented as https://github.com/Rickgg/ESP-Dmx already does this on ESP8266 in Arduino. Also the define in `uart.h` suggests that this shoud be possible (unless there is some reason why Lua implementation would not support this particular bitrate from all those available).

devsaurus, 2016-02-14:
Agreed, there's no obvious reason why the definition from `app/include/uart.h` shouldn't be picked up by the platform code. #856 suggests that it would be best to actually verify the HW support for this setting with a scope.
I'll look at it during the next days.

devsaurus, 2016-02-16:
The related PR has been merged to `dev`. Please check whether it resolves your issue.

PR #1056 (2016-02-15): Enable UART baudrate 256000 in platform layer.
Fixes #1045 and corrects doc wrt baudrate 3686400.
closed 2016-02-19 by devsaurus