nodemcu/nodemcu-firmware #1046: Is there a max size for the Irom0text.bin?

jfollas, 2016-02-13:
This issue originally surfaced when I had #DEVELOP_VERSION enabled to do some debugging (https://github.com/nodemcu/nodemcu-firmware/issues/976). I already need a lot of different modules, and with the additional debugging, it pushed the build size of the 0x10000.bin over 0x6c000 (442,368 bytes) in length. Note that this is for an ESP-12, so flash storage space was not a concern.

Upon reset, different behavior was observed (Panic-Reset loops, spitting out 0xFE characters forever, or just repeated resets with no Panic messages, etc). But the consistency was in the fact that large builds refused to work. (My toolchain is using 1.5.1 SDK)

My question for the contributors here: is there something magical about that 0X6C000 length in general, so that builds simply will not work if it is larger? Or could there be an issue with the linker scripts, or IRAM size, etc?

I discovered in Espressif's 2A documentation (http://bbs.espressif.com/download/file.php?id=1074) that in SDK 1.5 that the flash map is as follows (for the 4096 KB flash size, which the ESP-12 module has):

```
00 0000 - 00 FFFF = Flash.bin     (max 64 KB)
01 0000 - 03 BFFF = User Data     (176 KB + any leftover in flash.bin's 64 KB space)
03 C000 - 03 FFFF = User Param    (16 KB)
04 0000 - 0F FFFF = Irom0text.bin (max 768 KB)
0C 0000 - 3F BFFF = User Data     (3056 KB + any leftover in Irom0text.bin's 768 KB space)
3F C000 - 3F FFFF = System Param  (16 KB)
```

This seems contrary to what is currently being created and flashed (i.e., NodeMCU currently flashes Irom0text.bin to 0x10000). Is this just a red herring, and it doesn't matter where files actually get flashed? 

When we get into toolchain configuration and linker scripts, I quickly get over my head, so that's why I'm bringing the issue to the folks here with more experience. I'm not sure how to troubleshoot the issue at this point.

TerryE, 2016-02-14:
This is really falling outside the scope of this issues list unless you reframe this as a request for a specific change to the how we document this in our standard documentation. (See #719  and our support documentation.)

I keep my builds minimal for testing and use the reduced pin modules as these use QIO rather than DIO, and this means that the flash times are roughly 3-4x quicker.  I only turn on debugging on a per module basis for the modules that I am debugging.   The 768Kb comes from the instruction address space only being 20-bit on the ESP8266 and the first 256Kb is used by the iram0 and ram address spaces.  See the documentation which give a link to my esp826.com wiki article on how to build a VirtualBox VM for doing dev.  You can run this on M$, OSX or Linux, and it takes about 30 mins with a decent internet connection, but this isn't the forum to have these discussions.

closed 2016-02-14 by TerryE

jfollas, 2016-02-14:
FWIW, though I already had a linux-based build environment running on a Raspberry Pi 2, I stood up a second one on an Ubuntu VM in Azure, and the binaries that it produced don't seem to have the same issue! 

So, there's something strange happening - maybe related to ARM in the RaPi2 - but for the moment, I'm satisfied with having a way to build the firmware images that include all of the modules that I need for my project.

TerryE, 2016-02-14:
@jfollas as you are a contributor, I'd be happy to have a side dialogue on esp8266.com or you can PM me there and we can swap email details.  It's just that if we do it here, then we open the flood gates :smile: