nodemcu/nodemcu-firmware #1047: How to share the FRC1 timer between modules/drivers?

pjsg, 2016-02-13:
The FRC1 timer is (apparently) unused by the SDK and available for use by application code. It provides a precise timing mechanism that calls into application code at interrupt level. The timer can be configured to use either the NMI or a regular interrupt. These differ in that the NMI cannot be masked and is of the highest priority.

There are several applications for this timer:

- PWM driver. This builds tables that give timing information as to when the GPIO outputs should be switched. It uses the FRC1 to do this. [I suspect that it does not take into account the time between getting the interrupt and when it sets the next timeout]

- Performance monitor. This interrupts at regular intervals to record the value of the program counter so that a histogram can made,

- Instrument gauge stepper motor driver. This needs precise timing to output the correct pulsetrain. 

There appear to be two approaches to the sharing of the timer:

- Explicitly take ownership of the timer and use it for amount of time needed. Then release the timer so that another module can use it.

- Build a shared timer infrastructure that would (I guess) be similar to the software timers except that the callbacks would be at interrupt level.

It is clear that the first approach is much simpler, but it does mean that only a single module can get timer functions at a time.

Thoughts?

TerryE, 2016-02-13:
Phillip, my own view is that the ESP8266 is an excellent IoT processor with WiFi capability.  It is a mistake to try to over-stretch its capability.  Trying to make FRC1 a shared resource.  My inclination is to keep this all as simple as possible.  Your option (1) represents a step improvement over the current setup, so I think that that is the best approach for now: you have one available H/W timer; you can attach one H/W timer-based library to it.  If you need to support more than one such library, then use multiple ESP8266s

pjsg, 2016-02-14:
I can certainly relate to that. I'll brew something up....

TerryE, 2016-02-14:
Philip, before you cut the code, any chance of drafting a spec of what you suggest in terms of the functions that you propose to add to platform?  E.g. I assume that you will have some timer init() and close() with a parameter set to bind the ISR and set up the timer frequency and type.

pjsg, 2016-02-14:
The basic idea is that FRC1 is owned by zero or one specific module at a time. Each of the timer_functions gets an additional argument which is the claimed owner. If the claimed owner is not the actual owner, then the operation fails. 

The owner is represented as an os_param_t. One way of ensuring unique values between modules is to make these pointers to some private data/code. 

All these functions (except for _init) can be called at interrupt level.

All this code is heavily based on [hw_timer.c](https://github.com/willemwouters/ESP8266/blob/master/projects/modules/modules_1_5_0/driver/hw_timer.c)

## bool platform_hw_timer_init(os_param_t owner, FRC1_TIMER_SOURCE_TYPE source_type, u8 req)

This sets up the timer according to the parameters and claims it (if not already claimed). If some other module has already claimed it, then this returns false, other true.

## bool platform_hw_timer_set_func(os_param_t owner, void (* user_hw_timer_cb_set)(void))

This sets the callback to be the supplied function. The timer must either have been unowned or owned by the current module. It returns false if the operation fails.

## bool platform_hw_timer_arm_us(os_param_t owner, uint32_t us)

Arms the timer for the supplied number of microseconds. Again returns false if the owner is not correct.

## bool platform_hw_timer_arm_ticks(os_param_t owner, uint32_t ticks)

Arms the timer for the supplied number of timer ticks. Again returns false if the owner is not correct.

## uint32_t platform_hw_timer_us_to_ticks(uint32_t us)

This returns the number of ticks for a given number of microseconds. This operation is expensive (it involves multiplication and division) and should be avoided in the fast path.

## bool platform_hw_timer_close(os_param_t owner)

This releases the timer (if it was owned) and makes it available for other modules to use. Returns the 
standard true if it worked.

devsaurus, 2016-02-14:
Some comments based on my experience with this timer:

> bool platform_hw_timer_init(os_param_t owner, FRC1_TIMER_SOURCE_TYPE source_type, u8 req)

Espressif's examples call the parameter that selects autoload vs single shot `req`. Not very intuitive IMO. Why not `autoload` for WYSIWYG?

> bool platform_hw_timer_set_func(os_param_t owner, void (* user_hw_timer_cb_set)(void))

`ETS_FRC_TIMER1_INTR_ATTACH()` can provide an `arg` parameter to the ISR. The above prototype won't support that. But since `ETS_FRC_TIMER1_NMI_INTR_ATTACH()` *doesn't* provide `arg`, it's probably better to skip it altogether like in your proposal and save the headaches involved.

Excellent sketch, :+1: from my side.

TerryE, 2016-02-14:
My instinct is to keep the arg parameter, and to avoid the NMI as there be demons and we don't know the constraints of execution at NMI (e.g. can you safely post a task at NMI level).  However running an NMI autoload is almost certainly a lot better for system stability than doing what some libraries currently do, which is to mask interrupts for extended periods.

pjsg, 2016-02-14:
Keeping the callback 'arg' parameter is not a problem, and I'll make the argument names more explicit too.

pjsg, 2016-02-14:
See PR #1054 as the first step....

TerryE, 2016-02-14:
Good, but there's something flawed about the PMW implementation, IMO.  Far safer to use something like the PCA9685 (Banggood do a good 16 channel I2C module which has a 0.1" pin pitch for £3.50), so you are welcome to take on sorting out the PMW module :relieved:  I'll comment on your draft at the PR.

pjsg, 2016-02-15:
The pwm implementation is weird. There are straight bugs in it. 

My testing was derailed for a long time as pwm.setup(10, 1000, 512) causes a lockup on my board. It seems that this (shared) pin is special in some way. I'll see if I can fix that as well....   I was using 10 as it was convenient on my board to put and LED there....

pjsg, 2016-02-15:
Argh -- the pwm code is messy. Anyway it is a lot better now.

See PR #1057 for the full gory details.

TerryE, 2016-02-16:
Will take a look :+1:

pjsg, 2016-03-10:
This was merged.....

closed 2016-03-10 by pjsg