nodemcu/nodemcu-firmware #1052: I2C Rework

TerryE, 2016-02-14:
After I finish the GPIO module, I've got a few small bits of backlog like the encode module that I want to flush through as commits, but I am thinking of doing a higher level I2C implementation to make it  lot simpler to write I2C based drivers -- possibly along the printf / scanf lines. 

This may seem an odd idea but I2C drivers at the moment are very cumbersome since a lot of API calls are involved, and so take up a lot of code space.  In reality the drivers spend most of their time in delay loops, so parsing a format-style encapsulation can be lost in these time windows.  This would also mean that coding up such drivers as Lua modules would be just as simple and fast as having to drop down into a C module.

This is just a place-holder for feedback and comment, before I start this mini-project.

IVI053, 2016-04-23:
Will this include adding the functionality of running as slave device?

TerryE, 2016-04-23:
Running the ESP8266 architecture as a slave is a lot more challenging as the host controls the clock and this would need time critical ISR-based timing. This is an my medium term more of a wish list priority, because I need to sort out a quite a few other PRs first.

robstarling, 2016-05-05:
Do you plan to support clock-stretching? It appears that the current implementation completely ignores it. This is important for slower I2C slaves. I believe i can help implement it.

jsaily, 2016-05-18:
robstarling: Connecting with a slow I2C slave is exactly my problem too! I can easily connect to a PCF8574 I/O expander but my PIC18F2520 slave processor is not even giving ACK to the ESP8266 running NodeMCU. I would only need a basic NodeMCU build with file, GPIO, net, node, timer, UART, WiFi and slow I2C modules. 

I was wondering if I could hack the I2C module to increase delays everywhere, and compile with Docker build tool? https://hub.docker.com/r/marcelstoer/nodemcu-build/

Xasin, 2017-05-27:
@jsaily 
And, did things work out? If so, I might be interested in doing the reverse - speeding things up a bit to communicate at faster speeds with my Arduino.

TerryE, 2017-06-18:
Just a codocil on this one: to be honest, on reflection reworking I2C isn't a high priority for me now.  Yes, there would be lots of nice-to-have items here, but the current I2C implementation is pretty solid and usable for everything I've used it for, so I am inclined to close this issue.

Xasin, 2017-06-21:
OK for me - A lot of optimization could be squeezed out by storing all variables to be sent in a single-layer array, instead of a 2D one that had to be "squeezed flat" first.
Brought send time down from ca. 12ms to 5ms, enough for me.

closed 2017-07-01 by TerryE