nodemcu/nodemcu-firmware #1058: Rotary Switch module

pjsg, 2016-02-16:
This is a proposal to add a module that can decode a standard rotary switch -- typically found in automotive audio systems. These are available at all the standard online stores for a [buck or so](http://www.aliexpress.com/item/360-Degree-20-Detents-5Pin-Digital-Rotary-Encoder-Momentary-Switch/32219946924.html). They have a quadrature output on two pins and a momentary press button.

These switches can be connected using three GPIO pins. 

The approach is to enhance the GPIO module to allow other modules to handle gpio interrupts as well. Then this module will register and handle the interrupts for these three pins. The pin values are then converted into the rotary (or press) motion and the absolute values are maintained by the module. Changes are then sent to a callback via the new task mechanism.

dvv, 2016-02-16:
+1

I would however notice interrupt-driven mechanical buttons may require good debouncing.
FYI, here I use customizable polling https://github.com/dvv/nodemcu-thingies/blob/master/buttons.lua

pjsg, 2016-02-16:
Indeed -- debouncing the press button is important. For the quadrature outputs it doesn't matter so much as each bounce just moves the position one click rather rapidly. The standard switches have 20 (or 16) detents and there is a full 4 quadrature advances between each detent.

pjsg, 2016-02-17:
This is the interface: [rotary.md](https://github.com/pjsg/nodemcu-firmware/blob/rotary-switec/docs/en/modules/rotary.md)

pastukhov, 2016-02-17:
Please add to documetation some links to a compatibile hardware (alexpress, ebay, seedstudio)

dvv, 2016-02-17:
Wonder if `PRESS`/`LONGPRESS` etc. logic deserved to be put in the module.

mrkale, 2016-02-17:
And there is the module located?

pjsg, 2016-02-17:
The module does not yet have a pull request as it is waiting for another PR #1023 to be merged.

I like the idea of `LONGPRESS` and I'll add that before I submit the PR. What do you think is the time boundary between short and long press?  I'm thinking somewhere between 0.5 and 1 second.

 I suspect that the following would happen (in terms of the callbacks):

On the initial press: `PRESS`
After the time internal: `LONGPRESS`
When released: `RELEASE`

I'll also update the docs with links to the hardware.

dvv, 2016-02-17:
I believe your times right. May be options to setup (to vary between channels)?
May be even to borrow the logic of "mouse" events? E.g. PRESS, RELEASE, CLICK; PRESS, RELEASE, {shortly}, PRESS, RELEASE, DBLCLICK; PRESS, {long}, LONGPRESS, RELEASE.
These three types would poke the majority of needs, I believe.

pjsg, 2016-02-18:
I added the links to parts.

However, the long press stuff is rather more tricky than I had anticipated. The reason is that all the other events are triggered directly as the result of a GPIO interrupt. The LONGPRESS event and the CLICK/DBLCLICK events are all triggered on timers. For this sort of timing, I'd have to use a hardware timer -- but there is only a single hardware timer and that is used (more importantly) for other devices -- such as PWM output (or another new module that I've been working on).

What I could do would be to deliver the press/release timestamps so that the lua code could figure out how long the press was.

TerryE, 2016-02-18:
> For this sort of timing, I'd have to use a hardware timer

Why? Why not just use a SW timer? Surely epochs of 0.5-1 sec are ideal for this.

marcelstoer, 2016-03-01:
Fixed by #1067.

closed 2016-03-01 by marcelstoer