nodemcu/nodemcu-firmware #1065: wifi.sta.eventMonReg previous_State is always "nil"

trickkiste, 2016-02-19:
This example from the doc, does actually NOT work.

previous_State is "nil" for all events.

--register callback: use previous state
wifi.sta.eventMonReg(wifi.STA_CONNECTING, function(previous_State)
    if(previous_State==wifi.STA_GOTIP) then 
        print("Station lost connection with access point\n\tAttempting to reconnect...")
    else
        print("STATION_CONNECTING")
    end
end)

Using dev tree from cloud builder tool.
And also true for master.

devsaurus, 2016-02-19:
A quick code review confirms this mismatch between API doc and actual code implementation. Thanks for reporting.
It seems that #556 got lost in time and space. I'll have a look at how to re-apply this.
@dnc40085 do you expect any issues?

dnc40085, 2016-02-20:
@devsaurus Wow that's weird, I knew I added that.
I'm currently reworking the #1018 to take advantage of the new tasking interface, If you'd like I can add a fix for this before I commit the changes to the online repo later today.

devsaurus, 2016-02-20:
Thanks @dnc40085, this looks like an all-in-one solution :smiley: 

@trickkiste as a short term mitigation of your problem you could look at [`devsaurus/wifi_event_cb`](https://github.com/devsaurus/nodemcu-firmware/tree/wifi_event_cb). I retro-applied the original commits which added the missing call parameter. It's rebased onto current `dev` but untested.

devsaurus, 2016-03-17:
Fix applied to code base, thanks @dnc40085.

closed 2016-03-17 by devsaurus