nodemcu/nodemcu-firmware #1069: net.socket:send() fails for some clients, when the length of the sent payload is small

etisab, 2016-02-20:
Hello,

I am serving very small web pages from my ESP. I would like to accommodate as many devices as possible, especially mobile. I've come across an issue that so far affects only iPhone 6 Plus. 

When serving a small page, the connection fails with "connection lost" reported on the device, but when serving pages close to the 1KB limit, the page is served as expected. Duplicating code at bottom.

Details:
- All pages are served correctly on Samsung S3 and Nexus 6, as well as any desktop browser.
- Pages smaller than ~500KB report "connection lost" when using iPhone 6 Plus. Verified not working with latest Safari, Chrome, Firefox 
- Using latest dev branch, commit dc905f080b3b54
- In all cases the iPhone makes 3 requests to '/' in quick succession, so I'm suspicious there's a race condition of some kind, or the ESP breaks under the load
- Have not tested other iDevices since I don't have access to them yet

```
wifi.setmode(wifi.STATIONAP)

local accessPointConfig = {}
accessPointConfig.ssid = "myhotspot" .. node.chipid()
accessPointConfig.pwd = "password"
accessPointConfig.max = 1
accessPointConfig.auth = wifi.AUTH_OPEN

wifi.ap.config(accessPointConfig)
wifi.ap.dhcp.start()

-- Create the httpd server
svr = net.createServer (net.TCP, 30)

-- Server listening on port 80, call connect function if a request is received
svr:listen (80, function(conn)
   conn:on ("receive", function (conn, req_data)
      print("REQ DATA: " .. req_data)
      -- this call fails for iPhone 6 Plus
      -- conn:send ("<!DOCTYPE HTML> <h1>Hello NodeMCU!</h1>")

      -- this call succeeds if the string sent is long
      sendIndex(conn)
   end)

   conn:on ("sent", function(conn)
      -- Close the connection for the request
      conn:close ( )
   end)
end)

function sendIndex(conn)
   indexhtml_fails = "AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA"
   indexhtml_succeeds = "AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA\nAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA"
   conn:send(indexhtml_succeeds)
end
```

marcelstoer, 2016-02-21:
I don't completely understand which lines of code are actually active in the failure case.

However, never forget that you're working with a fully asynchronous firmware. You shouldn't ever have multiple consecutive calls to `conn:send`: http://nodemcu.readthedocs.org/en/dev/en/modules/net/#note.

DesperateProgrammer, 2016-02-21:
Send a valid http response header and iOS will accept it regardless of size.

I dont think this is a nodemcu issue.

TerryE, 2016-02-21:
+1 on @DesperateProgrammer 's point.  Both replies are invalid  both HTTP and HTML responses which means that the browser responses are indeterminate.  If @etisab can't provide an example with two responses which comply with both HTTP and HTML RFCs one of with fails and one which succeeds, then this is definitely not a firmware issue and we will close this.

devyte, 2016-02-21:
I didn't use this example, but with my nodemcu-platform (a nodemcu-httpserver variant), a simple "Hello NodeMCU!" html works correctly in iOS safari on an iphone 5, android chrome, android opera, android firefox, windows IE, windows Chrome, windows Firefox. I didn't test others.

I'd also say this not a issue with the firmware.

etisab, 2016-02-21:
@DesperateProgrammer You are right, adding a status line per rfc2616 resolves the issue. I hadn't considered this case since the same browsers on all other platforms accept it. Thanks everyone for your input.

closed 2016-02-21 by etisab