nodemcu/nodemcu-firmware #1073: Doc in PDF ?

ghusson2, 2016-02-22:
Hello,
I am traveling for work and this time is good form me to spend some time over my NodeMCU related projects. And an online documentation is not adapted. It there a way to download a PDF from Read the Docs ? In features I see : PDF Generation, but I can't figure out how to get this...

Thanks.

ABonner, 2016-02-22:
The NodeMCU documentation is using the `mkdocs` build engine to generate the HTML documentation you see online. [Per the readthedocs.org site](https://read-the-docs.readthedocs.org/en/latest/builds.html#mkdocs), this engine does not support PDF generation:
> As MkDocs doesn’t support automatic PDF generation, Read the Docs cannot create a PDF version of your documentation with the Mkdocs option.

The project could theoretically use the `Sphinx` build process with support for `.md` files but that should be a separate issue that is opened if someone is interested in doing so.

TerryE, 2016-02-22:
@ghusson2 It would be nice but Adam has already given the issues with this.  We just don't have the effort to implement this at the moment.   

If you keep a local copy of the git repo on your PC / laptop, then it is only a few commands to download the python engine for mkdocs.  Even though I run Ubuntu on my laptop, I still run my nodeMCU dev environment in its own VM and I can start mkdocs in it.  Port 8888 is my mkdocs server and I've mapped this to localhost port 8888, so opening up `http:localhost:8888/en/`brings up the documentation for my working branch.  OK, I do this because I am sometimes writing the docs, but I also use it for double checking stuff.  

There is no reason why you couldn't do the same. 

Given that we have no immediate plans to do this. I will close this issue.  Sorry.

closed 2016-02-22 by TerryE

marcelstoer, 2016-02-22:
@ghusson2 Also have a look at https://github.com/jgrassler/mkdocs-pandoc and https://github.com/rtfd/readthedocs.org/issues/1939

ghusson2, 2016-02-22:
Hello,
Thank you guys for those tips. I will first try a recursive wget (simple/fast to try). If it is not satisfying, I will try mkdocs and mkdocs-pandoc.

truhlikfredy, 2016-10-11:
for me pretty well the works (without a theme which I like for printing)

```
for f in *.md
do
 echo "Converting $f"
 markdown $f | htmldoc --cont --headfootsize 8.0 --linkcolor blue --linkstyle plain --format pdf14 - > $f.pdf
done

pdftk *.pdf cat output _combined.pdf
```

I run this once in docs/en/ and then in docs/en/modules and I have 2 PDFs with everything I want. I could combine both of them bu often I just want the modules only anyway.