nodemcu/nodemcu-firmware #1075: file.read(x) may read less than x

marcoskirsch, 2016-02-22:
See the following code:

      -- create 4 KB file.
      file.open("test", "w+")
      for i=1, 1024, 1 do file.write("bug!") end
      file.close()

      -- Read more than 1 KB, see if the amount read is what's expected.
      file.open("test", "r")
      x = file.read(2048)
      if x then
         print("Data size expected 2048, is: ", #x)
      else
         print ("WTF")
      end
      x = nil

The output is:

      Data size expected 2048, is: 	1024

This is surprising behavior and it doesn't match documentation (and maybe it boils down to fixing documentation). My build info:

      NodeMCU custom build by frightanic.com
      branch: dev
      commit: fcb14a33b3aa17ba4173d2e70bc6e1ca1e036a98
      SSL: false
      modules: crypto,file,gpio,net,node,tmr,uart,wifi
      build 	built on: 2016-02-06 03:55
      powered by Lua 5.1.4 on SDK 1.5.1(e67da894)

ABonner, 2016-02-22:
Reads are limited to the `LUAL_BUFFERSIZE` that is chosen at compile time. It is 1024 by default. I'd actually hit this before as well and added some updates to the docs to clarify for the future.

TerryE, 2016-02-22:
@ABonner Adam, as an aside the file code for this isn't very efficient.  I might just tweak this, but I would leave the actual observed functionality unchanged.  The reason that this returns a "bytes read" is that it read can return less, so :+1: on addressing this through documentation.

closed 2016-02-22 by TerryE