nodemcu/nodemcu-firmware #1077: Create Issue and Pull Request templates

marcelstoer, 2016-02-22:
As per https://github.com/blog/2111-issue-and-pull-request-templates create `.github/PULL_REQUEST_TEMPLATE.md` and `.github/ISSUE_TEMPLATE.md.md`.

jmattsson, 2016-02-24:
Ooooh, nice feature! Sounds like we could preempt a bunch of 719 stuff this way!

marcelstoer, 2016-02-24:
Exactly.

> 719 stuff 

Shouldn't this be "1010 stuff" by now...nudge, nudge :wink:

jmattsson, 2016-02-24:
Yeah, I still have that issue open in a tab. I'm flat out with work stuff at the moment and haven't even got time & energy to keep my github notifications list empty. Another couple of weeks and the insanity should reduce. I hope.

Contributions of content of 1010 happily accepted - you have my email addy ;)

marcelstoer, 2016-04-13:
Before I create a PR let's discuss the following proposals:
- issues https://github.com/nodemcu/nodemcu-firmware/wiki/%5BDRAFT%5D-GitHub-issue-template
- PRs https://github.com/nodemcu/nodemcu-firmware/wiki/%5BDRAFT%5D-GitHub-PR-template

Feel free to edit those directly.

A note about the last line on the PR template. I have the feeling that PRs are reviewed quite thoroughly but usually there's little discussion whether we want to "adopt" new modules (for example) _at all_. We kind of assume full responsibility once a module has landed in this repository. By maintaining a list of supportive votes right in the description of the PR we'll get a rudimentary assessment of the "usefulness" of the contribution. In the past you'd have to read and interpret a few, or a few dozen, comments to arrive at that conclusion. My intention is not to say "After all the refinements I'm finally giving this a thumbs up" but more "Regardless of the implementation I'd like to see this added to NodeMCU".

jmattsson, 2016-04-14:
I made some minor tweaks.

Going on past template experience for bug reports, I'm not sure this will work as well as we hope, but it's worth a shot.

marcelstoer, 2016-04-14:
I have very little hopes but it certainly won't make it worse so, yeah, worth a shot.

TerryE, 2016-04-14:
I think that this type of template should only apply for bugreps from non-contributors to help weed out chaff, but to be honest I can't see myself using them; the bureaucracy of all this is just too cumbersome for me. 

 A template should be a checklist for guidance, not a straight jacket.

marcelstoer, 2016-04-14:
> I can't see myself using them

GitHub just prefills the textarea with the template content. Anyone is free to delete all of that.

> A template should be a checklist for guidance, not a straight jacket.

Yes, that's the intention.

TerryE, 2016-04-14:
Thanks Marcus  realise this, but let's see how many posters conform.  A lot of one-off reporters can't even be bothered to use Markdown.

jmattsson, 2016-04-26:
Merged. Let's see how this pans out...

PR #1257 (2016-04-26): Add issue and PR templates
Fixes #1077, as discussed there.
closed 2016-04-26 by jmattsson