nodemcu/nodemcu-firmware #1078: "GPIO rework" merged to dev on 18 Feb broke GPIO triggering

NicolSpies, 2016-02-22:
@TerryE @jmattsson 
I constantly follow all the comments on Pulse and appreciate what I am learning from you. 
I am developing full time using custom NodeMCU firmware builds.

1. I have verified, using the same test code utilizing **GPIO trigger functionality**, working on the dev commit merged by TerryE on 17 Feb, **stopped working** on merged dev commits made on **18 Feb**.
 
2. I have also verified on the latest build made today that it is still not working.
 
3. It appears to be related to something that changed related to the "Merge pull request #1061 from TerryE/dev-task-if" committed by jmattsson (Add New Tasking I/F and rework GPIO, UART, etc to support it)

TerryE, 2016-02-22:
@NicolSpies Hi Nicole, I am a little confused. You should really quote the short SHA-1s for avoidance of doubt, for example I take you to mean the "commit merged by TerryE on 17 Feb" as 49733f6, and the "merged dev commits made on 18 Feb" as 1afa72d.

The way we do commits to the dev repo is to use a "two person" rule so this second commit was by merge by Johny of the first.  I've just retried by gpio test stub on a later dev commit and this works fine. 

So Nichol, can you please reference which dev commit that you working against and provide (say as a gist) the test that you are using with description of the test and a paste of the outputs, so we know what to chaise down.  Thanks.

NicolSpies, 2016-02-22:
@TerryE  Hi Terry, I apologize for the confusing report. Commit https://github.com/nodemcu/nodemcu-firmware/commit/19471f54adf095ecc7b2dd6515592b60fe0bf30c is working fine. The next build I made the following day of commit https://github.com/nodemcu/nodemcu-firmware/commit/1afa72d53a6a1b9acc40c8bf8537ff7fdbbb0c87 manifests the problem.

My test code uses the i2c module to read button interrupts from a MCP23008 i2c GPIO port expander.
I configure with `gpio.mode(2, gpio.INT)`. In the working commit `gpio.trig(2, "low" , function() readi2c() end)` triggers the callback function.

In the dysfunctional commit the callback function is not triggered on a button push. After a few seconds an unprotected error creates a panic and the module resets.

I can try and capture the "garbage" when it reboots if it will be helpful. Thanks

TerryE, 2016-02-22:
On thing that I did notice in testing is that low or high continually fire the interrupt.  Have you though of triggering on the falling edge -- that is to use "down" instead of "low".  Failing that stay with your working commit, until we bottom this problem.  Post a gist of your relevant code would be useful.

NicolSpies, 2016-02-23:
Even the working commit "low" fires the interrupt continually and "down" does not fire it at all. I "managed" to work around the "continual low firing" in my "readi2c" function called.

So, the plot thickens. I am starting to believe that the problem exists in the working commit as well. It might be that my "work around" does not work in the later commit due to something fixed in the later commit.

So my latest take on what I am experiencing is that the "down" edge trigger seems suspect. I will test my code against my collection of commits going back to September last year. This will hopefully find a commit where "down" stopped working.

NicolSpies, 2016-02-23:
Further feedback. This is confusing.

1. Test code used : https://gist.github.com/NicolSpies/a7468ca6939a4839d3ba#file-gistfile1-txt
2. All commits from  https://github.com/nodemcu/nodemcu-firmware/commit/1afa72d53a6a1b9acc40c8bf8537ff7fdbbb0c87 forward to date crashes the test code.
3. In all commits from https://github.com/nodemcu/nodemcu-firmware/commit/19471f54adf095ecc7b2dd6515592b60fe0bf30c going back to September last year does not crash the test code.
4. The working test code behaves as follows:

4.1 When using `gpio.trig(pin, "low" , function(level) readi2c(level)  end)` , the correct interrupt on the GPIO port expander is read by `local val = i2cRead(0x07)`. The value is printed as long as the iterrupt is present (as long as the button is pressed).
4.2 When I reset the ESP and uploads the test code again with `gpio.trig(pin, "down" , function(level) readi2c(level)  end)` for a downward edge interrupt, it reads correct once and then "latches" not reading afterwards again even if I reset the ESP.
4.3 The only way I can get it to work is to re-upload the test code with `gpio.trig(pin, "low" , function(level) readi2c(level)  end)`, then it runs again as in step 4.1.
4.4 If I now repeat step 4.2 it works once again and latches.

So, my limited and intermediate assessment is as follows:

1.  In the commit https://github.com/nodemcu/nodemcu-firmware/commit/1afa72d53a6a1b9acc40c8bf8537ff7fdbbb0c87 something changed in the GPIO code.
2. Before this commit triggering on "down" works once even when the ESP is reset and can only be restored to "work" when the test code triggering on "low" is uploaded. After the upload triggering on "down" will "work" once only before the process described in this point needs to be repeated.
3. Point 2 is very confusing, my only comment is something appears to "latch"

pjsg, 2016-02-23:
What puzzles me is how it could ever have worked -- but this indicates that I do not understand how the trigger on "low" actually works. 

It seems that the following signals affect whether an interrupt is generated:

* The level of the GPIO pin
* Whether there has been any change on the input pin since the last write to STATUS_W1TC
* Whether the interrupt mode of the pin has changed.
* The ordering of these events

I don't have a coherent mental model of how this works which would explain both the old behavior and the new behavior. My problem is that I don't see why the new code doesn't go into a hard interrupt loop with a low input (when trigger on low is set). I'm assuming that the next hardware interrupt happens very shortly after the interrupt is re-enabled on the pin (and this is in the interrupt handler). In master, the lua code would run before this happened and could (presumably) clear the interrupt. If it didn't it would get immediately re-invoked.

Confused.

[Apologies for the stream-of-consciousness, but I find it sometimes helpful to try and explain something...]

TerryE, 2016-02-23:
I think is worked previously because of an artefact  / bug in the previous driver because of the cb being delivered from within the driver rather than a separate task.  The current and nominally the previous algo had any triggers automatically self-arming so that they could repeat.  This makes sense to me to for edge-triggered interrupts but not for level triggered ones, because if the the pin is held at the level then the trigger fires machine-gun mode and even though the task delivery is only re-enabled as part of the trigger task, I was still getting over 10 tasks per mSec for a level based trigger.  

What makes more sense to me is that a level-based trigger is only delivered once.  The cb can then do the necessary I/O processing (and thereby clear the level)  before reissunig the gpio.trig() to reenable the next trigger.

NicolSpies, 2016-02-23:
Hi, thanks for your reflection. I trigger on the INT line of the GPIO port expander that goes active low when an input change on the port expander is sensed. Because I cannot catch the edge going low on the INT, I trigger on the low level. When this happens the callback function reads the data on the i2c bus. On every read the port expander is "cleared". With the INT still low, it is  triggered again and the value is read again until I let go of the button INT goes high and the calling of the i2c function stops. 

This is a workaround because I couldn't trigger on "down" for some reason. This situation is present back to commits made last year in September and might be a different problem stumbled apon.

My workaround was fine until the commit made on 18 Feb where the GPIO was reworked having the effect that my workaround stopped working.

I hope this clarifies a bit more

pjsg, 2016-02-23:
I'm going to try and conduct some experiments. To me, what *ought* to happen for a level based trigger is that the triggered function should be called repeatedly until the triggering condition goes away.

I'm going to see if I can write some tests (with a couple of GPIO pins wired together so that I can trigger interrupts and the like).

TerryE, 2016-02-23:
> what ought to happen for a level based trigger is that the triggered function should be called repeatedly until the triggering condition goes away.

Yup, but the ISR fire continuously.  Bad news.

pjsg, 2016-02-24:
Not if it is disabled until the triggered function returns.....

jmattsson, 2016-02-24:
Yeah the interrupt needs to be masked until the high-level handler has done its thing.

pjsg, 2016-02-24:
I have a PR #1082 which (I hope) addresses this. Interested in feedback....

TerryE, 2016-02-24:
I'll take a look when I am properly awake and not just killing an insomnia interlude. :sleepy:

NicolSpies, 2016-02-24:
@pjsg I will test on my hardware as soon as the commit has been merged to dev on frightanic

pjsg, 2016-02-24:
@NicolSpies Unfortunately we really want feedback before merging the PR. Can you build my branch like you built the other versions to narrow down the problem commit?

NicolSpies, 2016-02-24:
@pjsg I would really like to give that feedback, unfortunately all my previous builds are done using the http://nodemcu-build.com/ facility. I must confess I do not know how to do it in another way yet. I have used my time to put to use what the group have been developing. So, if I can be so forward to ask for a build to be made :blush: and emailed to me, I promise to spend time on how to do it myself.

TerryE, 2016-02-25:
OK,
I think that I've worked this one out and I screwed up.  In terms of the GPIO ports `platform_gpio_intr_init()` does two things:

-  It sets the corresponding bit in GPIO_STATUS_W1TC_ADDRESS to clear that interrupt
-  It sets the interrupt type (e.g. low) for the pin.

The old code wrote to the `GPIO_STATUS_W1TC` register to clear down the interrupt, called the cb then did a `gpio_pin_intr_state_set(GPIO_ID_PIN(pin_num[pin]), type)` to reset the pin type.

My code wrote to the `GPIO_STATUS_W1TC` register t posted the cb task then did the `gpio_pin_intr_state_set(GPIO_ID_PIN(pin_num[pin]), type)` to reset the pin type, and the task itself was done sometime later, _after the pin had been reset_

We just need to move `gpio_pin_intr_state_se()t` into `gpio_pin_intr_state_set()` after the lua_call()` .

pjsg, 2016-02-25:
That is pretty much what I proposed in my PR to fix the problem..... #1082

TerryE, 2016-02-25:
@pjsg, Philip, I agree that they are equivalent because (i) the pin_trig[] value can be derived from the pin type and this field is therefore redundant, and (ii) there is an issue over timing of reenabling the interrupt w.r.t the callback as I discuss in my inline comment: this is better done after in the case of a level trigger and before in the case of an edge one.  Bugger.

marcelstoer, 2016-02-25:
> I must confess I do not know how to do it in another way yet.

@NicolSpies it's actually quite easy, see http://nodemcu.readthedocs.org/en/dev/en/build/

NicolSpies, 2016-02-26:
@marcelstoer, thank you, I will immediately proceed

pjsg, 2016-03-10:
The fix was merged and should be in the dev branch,

PR #1082 (2016-02-26): Made level triggers work better than before
I hope that this PR fixes #1078. In my testing it makes level triggers behave in a reasonable way -- they get repeatedly called while the situation persists. However, there is only ever one callback queued for each pin at a time. 

I'm interested in whether this solves @NicolSpies problem.

@TerryE since you did the original big rework, I'd appreciate your input. I removed the pin_trigger stuff and just emulated it by disabling and re-enabling the interrupts on a per pin basis.

This code, when saved to a file and then dofile'd exercises some pieces  [... cut]
closed 2016-03-10 by pjsg