nodemcu/nodemcu-firmware #1081: Crash with current dev branch

pjsg, 2016-02-24:
I was trying to do something a bit evil:

```
     function asyncDelay(ms)
         local co = coroutine.create(function() 
           tmr.alarm(0, ms, 0, function()
             coroutine.yield()
           end)
         end)
         coroutine.resume(co)
     end
     
     asyncDelay(1000)
```

Anyway, it doesn't work. However, if you run this from the command line a few times, then the nodemcu locks up or traps. It doesn't seem to do this if I wrap it in a function. It seems a bit sensitive to the conditions.....

TerryE, 2016-02-24:
OK, I need to write up how coroutines and the SDK interact.  The bug here is how some modules snap and use gL, and what you are doing. 
-  In the SDK everything is broken into tasks. Tasks run to completion and return control to the SDK.  
-  You cannot guarantee which task is delivered next
-  Lua coroutining is cooperative so the only way that you should swap from L0 to L1, say, is by a yield or resume to swap control.

So

-  Tasks should _always_ run in L0 because that is the default and if you attempt to run a task in L1 you will die. 
-  Lx threads should always yield back to L0 which then runs out and returns control to the SDK, hopefully storing the coroutine in a persistent variable.
-  A subsequent task can then resume the coroutine.

This setup allows a coroutine to provide a background procedural threading with the main routine tasks proving the foreground actions.

Now consider your  code. This is the same as:
```Lua
     function asyncDelay(ms)
         local FN = function() coroutine.yield() end)
         local co = coroutine.create(function() 
           tmr.alarm(0, ms, 0,  FN)
         end)
         -- co is now suspended waiting to be kicked into life 
         coroutine.resume(co)  -- this starts co which books the alarm
         print(coroutine.status(co)) -- it has now run to completion and the thread is dead.
     end
     asyncDelay(1000)
```
The coroutine books an alarm task for delivery in `ms` mSec and completes returning control to L0, so the coroutine no longer exists and therefore the and ms delay later a separate task is executed:
```Lua
     FN() -- which should throw an error because it is running in L0
                        --  and you can't yield in L0.
```
except that the timer doesn't use `lua_State L = lua_getstate()` as it should but instead store the state (L1) at the time of the booking in `timer_t`, which no longer exists by the time that `FN()` is executed so Lua dies in a heap.  

This is just an example of #846 that I haven't got around to fixing.  On the long TODO list :(

devyte, 2016-02-24:
I was about to ask some questions, but reading #846 sort of answered the important ones.
Could I suggest adding all these explanations to the wiki somewhere for general access? Here as part of a discussion they will eventually be lost.
And thank you for the explanations @TerryE , lua does have a lot subtleties, both inside and out.

dnc40085, 2016-02-24:
@TerryE I went through the `tmr` module and made [these changes](https://github.com/dnc40085/nodemcu-firmware/commit/3cbd686fadf33f20ae3ab8caad7db28d86dfa58f) and ran @pjsg's code and got the following error:
>PANIC: unprotected error in call to Lua API (attempt to yield across metamethod/C-call boundary)

Was this the error that you were saying it should return?
Should I submit a PR or are there other changes that should be made?

pjsg, 2016-02-24:
I think that this is a good incremental improvement......

TerryE, 2016-02-24:
> Yes.

Try doing a `coroutine.yield()` at the `>` in the Lua interpreter (that is in L0) on the PC and you will be the same error.

TerryE, 2016-02-24:
> Should I submit a PR or are there other changes that should be made?

@dnc40085 I think that the first step is to use #846 to agree a proper programme of work and do a sanity check.  I am happy for you or Philip to do the work because we are all easily capable of this and I am a bit up to my eyeballs at the 'mo. This being the case, I think that Philip can probably close this one, since we understand this issue.   I'll post an analysis of this to #846.

TerryE, 2016-02-26:
This specific issues has been addressed by @dnc40085's recent PR #1083, but we are now aware of the underlying issue.  Thanks for the example, Philip.

closed 2016-02-26 by TerryE