nodemcu/nodemcu-firmware #1086: What is our practice for splitting large module files

TerryE, 2016-02-26:
This is a point that can up in #1018 and is worth an issue in its own right.  At the moment in general we stick to the practice of having one C file per library module in `app/modules` -- except in the case of modules which have their own folder tree such as `json`.  As a result the top level `app` directory is a mess. We have the following folder groups:

-  **Category folders**:  `driver`, `include`, `libc`,  `lua`, `modules`, `platform`, `user`
-  **Core libraries**:  `lwip`, `smart`, `spiffs`, `task`
-  **Module related**: `cjson`, `coap`, `crypto`, `dhtlib`, `http`, `json`, `mqtt`, `tsl2561`, `u8glib`, `ucglib` 
-  **Dead folders** which should be removed: `upgrade`,  `wofs`

This is just a mess.  I think that it would be a lot clearer if we moved to core modules into `core` (or some equiv name), and the module related as sub-folders of `modules`.  And the dead folders, we should just remove.

Thoughts?

jmattsson, 2016-02-26:
My understanding of the "module related" folders is that they're effectively 3rd party libraries we've imported, and to me it makes sense we keep them separate from the modules themselves.

I believe the `smart` folder should be categorised as "dead" - it's another remnant from 0.9.5/0.9.6.

I think currently our one "super module" is wifi, which is probably begging to be split up, somehow.

TerryE, 2016-02-26:
> My understanding of the "module related" folders is that they're effectively 3rd party libraries...

Yes many are and we should do minimum changes to the folder hierarchies and files, but what I am suggesting is that they should sit directory under `app` but should be moved under `modules` as they are module-specific.

As to the smart stuff, we still have `wifi.startsmart()` though how well this works isn't clear.  I don't know who uses it, and how well maintained it is.  Whether we keep it or not as @dnc40085 has pointed out, if we keep it then it's worth and a config define to make it optional as it takes up a lot of flash.  

We've got `task` and I was thinking of renaming this folder `core`, and using it to store any other core routines such as wifi and net helper modules if we decide to split these.

jmattsson, 2016-02-26:
We link fine without the `smart` directory - if anything I'd expect having it would cause us issues if it overrides the current SDK lib.

How's this for a rough draft of what we might do to structure things a bit better?
```
  app/
    |- core/
    |   |- lua/
    |   |- platform/ (platform.c, task.c, etc)
    |   |- driver/
    |   |- user/ (user_main.c, user_exception.c)
    |   |- utilities/ (for various useful things which don't fit into platform?)
    |   :
    |- core_libs/
    |   |- lwip/ (assuming the SDK lets us get away with moving it)
    |   |- spiffs/
    |   :
    |- modules/
    |   |- wifi/  (wifi.c, wifi_common.c, wifi_smart.c, etc)
    |   |- net/ (net.c, net_common.c, net_server.c, net_client.c, etc)
    |   |- gpio/ (gpio.c)
    |   :
    |- module_libs/
    |   |- coap/
    |   |- crypto/
    |   |- http/
    |   :
    :
```
I'm undecided whether I think each module should have its own dir or not. I like the simplicity of a single file for a module, so I think I'd like to have it both ways - single file for small modules and dir+multi-file for large modules.

TerryE, 2016-02-26:
My thought would be to hoist the contents of your suggested `core` back up to `app`, but this is what I am also thinking of.  The structure is a lot clearer.

TerryE, 2016-02-27:
Cf #1088 which tidies up a small part of this plus implements a suggestion by removing wofs and upgrade.

TerryE, 2016-03-02:
I also think that I agree with your suggestion to split module files when the size is unwieldy but to prefix all files with the module name (and any externs).  

Module names should be unique and lowercase ASCII alphanumeric.  (Do we permit "_" in module names as this is also the file-within-module separator?)
   
External folder hierarchies should only be used in the case of standard libraries.  This is definitely the case for `lwip`, `spiffs`, `u8glib` and `ucglib`, but some of the others are questionable.

TerryE, 2016-03-02:
I have just had a little further probe.  The following folders:

-   **`tsl2561` (541), `dhtlib` (388), `http` (727)** seem to be where module creators have decided to split their modules into a API call validation component in `app/modules` and an implementation in its own folder.  These are all small (the number in brackets is the no of source lines) and could be moved back into modules:.  
-   **`json` (577)** isn't used anywhere and seems to be a source version of the SDK's `libjson.a`.  This can be removed.
-  **`smart (844)`** seems to be an old nodeMCU-specific version of the SDK's `smartconfig`.  This code is referenced in `moodules/wifi.c` but only if `NODE_SMART_OLDSTYLE` is defined.  Given that this define isn't documented anywhere and specifically in `user_config.h`, it seems time to drop the `smart` folder and remove the  `NODE_SMART_OLDSTYLE` code from `wifi.c`.  @jmattsson Johny, you did some of this replacement work.  Comments?

-

jmattsson, 2016-03-02:
By the time I touched the smartwifi stuff that module was already using the SDK lib (libsmartconfig.a). Not having a test setup for Smart, I don't think the current NODE_SMART_OLDSTYLE has seen any testing apart from being compile tested. I haven't seen anyone complain that I broke anything in this area, so I think it we can pretty safely declare OLDSTYLE dead now and remove it.

Regarding `http/`, httpclient.{h,c} are 3rd-party by Martin d'Allens. I'd disagree with moving that into `modules/`.

TerryE, 2016-03-03:
OK, I'll do this shuffle as a low priority task in the next week.

TerryE, 2018-08-29:
This is now moribund and will be swept up -- if at all -- during ESP8266 + ESP32 code base integration.

closed 2018-08-29 by TerryE