nodemcu/nodemcu-firmware #1087: String formatting Python-style

TerryE, 2016-02-26:
String formatting is a pain in Lua, but In standard Lua you can do a simple hack:
```Lua
getmetatable("").__mod = function(a, b) if b then 
  return string.format(unpack(type(b)=='table' and b or {b})) 
  end;return a; end
```
and by magic `print("Pi is %7.5f" % math.pi)` works!

But you can't do this in nodeMCU because `getmetatable("")` is a rotable.  Would anyone object if I added this (the efficient C equivalent) _with a user config enabling define_ to lstring.c ?

mikewen, 2016-02-26:
+1 Wonderful.

jmattsson, 2016-02-26:
Sounds useful, though it'd be nice to have better handling of nil/false arguments than in the quick hack you listed.

TerryE, 2016-02-26:
If I did this, it would be in C. and I'll try to cover al of the realistic test cases, The only issue is that some variants extend the format specifier syntax to include named arguments, but I am not currently proposing to do this.

dvv, 2016-02-26:
Wonder what is wrong with `print(("Pi is %7.5f"):format(math.pi))`?

TerryE, 2016-02-26:
Nothing. really that what I do now, but you must include the brackets so its a little less intuitive and it generates more opcodes.  I just think

    "Pi is %7.5f" % math.pi

is more readable than

    ("Pi is %7.5f"):format(math.pi)

As I said, this would be only enabled if you defined something like `STRING_FORMAT_OPERATOR_ENABLED` in the `user_config.c`, so you don't need to use it.

dvv, 2016-02-26:
I see but Lua is Lua. We count from 1, and "" and 0 is truthy. And this is ok ;)
Personally, to my eye, in `"Pi is %7.5f" % math.pi` the second `%` looks like part of format string, and for `"Pi is %7.5f and Pi^2 is %7.5f" % { math.pi, math.pi*math.pi }` the situation gets worse (a redundant table construction).
Also, this issue may IMHO introduce module incompatibility.

TerryE, 2016-02-26:
> the situation gets worse (a redundant table construction)

Why ? it isn't and your have the same computation in the `():format()` variant.

> This issue may IMHO introduce module incompatibility.

(a) it can't if the app programmer doesn't configure this, and (b) even if this si done, String doesn't currently offer the mod operator so the only incompatibility would be if someone was using the mod operator to throw and error, and there are a lot of easier ways of doing this.

djphoenix, 2017-01-23:
Bump.
+1, _configurable_ feature sounds nice for me.

TerryE, 2018-06-23:
One for the  cleanup TODO list

TerryE, 2019-10-14:
It's such a PITA not having this, that I've added this tiny addition to #2192, so now
-  `s % v` is a synonym for `string.format(s, v)` and
-  `s % {a,b,c}` is a synonym for `string.format(s, a, b, c)`.

For example, this makes printing hex numbers so much easier  to write: `'0x%x' % n`

closed 2019-10-28 by TerryE