nodemcu/nodemcu-firmware #1089: HTTP GET request http.get() doesnt work with IP address

nonflammable, 2016-02-26:
this works:
`http.get("https://www.vowstar.com/nodemcu/", nil, function(code, data)
    if (code < 0) then
      print("HTTP request failed")
    else
      print(code, data)
    end
  end)`
this not works:
`http.get("http://120.26.37.149/nodemcu/", nil, function(code, data)
    if (code < 0) then
      print("HTTP request failed")
    else
      print(code, data)
    end
  end)`

i need use it with local IP address like 192.168.1.234

TerryE, 2016-02-26:
One for @vowstar Rui, I think.

marcelstoer, 2016-02-26:
Works for me.

```lua
http.get("http://120.26.37.149/nodemcu/", nil, function(code, data)
    if (code < 0) then
      print("HTTP request failed")
    else
      print(code, data)
    end
  end)
```
prints `Hello nodemcu`.

However, I can also confirm that the above code fails to access resources on hosts in the same local network for me (192.168.0.x). I was tempted to close this because it seemed unrelated to the HTTP module. Yet, if I connect using the net module's low-level methods it works:

```lua
sk = net.createConnection(net.TCP, 0)
sk:on("receive", function(sck, c) print(c) end )
sk:on("connection", function(sck,c)
    -- Wait for connection before sending.
    sck:send("GET / HTTP/1.1\r\nHost: 192.168.0.36\r\nConnection: keep-alive\r\nAccept: */*\r\n\r\n")
end)
sk:connect(80,"192.168.0.36")
```

pjsg, 2016-02-26:
I wonder if the espconn_gethostbyname actually handles IP addresses or not. The comments in the code seem to indicate that it does -- but the Espressif document does not mention that behavior. 

Safer to see if the host is numeric and not call espconn_gethostbyname

nonflammable, 2016-02-27:
http://120.26.37.149/nodemcu/
doesn't work - dev branche v1.5.1
http://192.168.1.x/ also :(

jfollas, 2016-02-27:
I was successful in having 1.5.1 hit Fiddler (HTTP debugging proxy) running on same 192.168.1.0/24 subnet.

```
GET http://192.168.1.25:8890 HTTP/1.1
Host: 192.168.1.25:8890
Connection: close
User-Agent: ESP8266
_____________

HTTP/1.1 200 Fiddler Generated
Date: Sat, 27 Feb 2016 15:51:20 GMT
Content-Type: text/html; charset=UTF-8
Connection: close
Cache-Control: no-cache, must-revalidate
Timestamp: 10:51:20.218

<!doctype html>
<html><head><meta http-equiv="Content-Type" content="text/html; charset=UTF-8"><title>Fiddler Echo Service</title> ...
```

However, it would not access files served by a web server on same machine, but I'm thinking that might be Windows Firewall related at this point.

jfollas, 2016-02-27:
I do find this interesting, and may explain bad behavior. When I enable logging for the HTTP module:

```
> http.get("http://120.26.37.149/nodemcu/", nil)

hostname=120.26.37.149
port=80
method=GET
path=/nodemcu/
DNS request
DNS found 120.26.37.149 120.26.37.149
Connected
Sending request header
GET /nodemcu/ HTTP/1.1
Host: 120.26.37.149:80  <--- NOTE THIS!
Connection: close
User-Agent: ESP8266

All sent
Disconnected
Chunk Size:13
Chunk Size:0
http_status=200
strlen(full_response)=203
response=Hello nodemcu<EOF>
```

Notice the HOST header uses the IP Address returned from DNS instead of the hostname. This will screw up any request where the server is expecting a hostname (i.e., any multi-tenant/shared web environment where requests are routed according to the provided HOST header)

marcelstoer, 2016-02-27:
> `http://120.26.37.149/nodemcu/`
doesn't work - dev branche v1.5.1

That's unfortunate but as I said, it does work for me, just tested again (`dev` branch a few weeks old, fcb14a33b3aa17ba4173d2e70bc6e1ca1e036a98). Did you also try with the `net` module code I also posted?

@jfollas what was that about the host header (the comment you deleted)? As I've shown there may indeed be something odd with the HTTP headers. Why else would the `net` module code work to retrieve data from a local subnet host while `http.get()` doesn't?

jfollas, 2016-02-27:
@marcelstoer: I had made a request with debugging enabled in order to print out the HTTP Request headers, and saw an IP address instead of DNS name in the Host header. Was convinced this was a problem, until I noticed the request URL was to an IP address. :-)  Hence the deleted post.

nonflammable, 2016-02-27:
http://nodemcu-build.com/builds/nodemcu-dev-8-modules-2016-02-26-16-45-57-integer.bin
modules: file, gpio, http, net, node, tmr, uart, wifi.

```javascript
wifi.setmode(wifi.STATION)
wifi.sta.config("MyWIFI","mypassword")
wifi.sta.connect()
wifi.sta.setip({ip="192.168.1.200",netmask="255.255.255.0",gateway="192.168.1.1"})
--must be static IP
tmr.alarm(1,1000, 1, function() if wifi.sta.getip()==nil then print(" Wait to IP address!") else
print("New IP address is "..wifi.sta.getip())
http.get("http://120.26.37.149/nodemcu/", nil, function(code, data) --nothing 
--http.get("http://www.vowstar.com/nodemcu/", nil, function(code, data) --200	Hello nodemcu
    if (code < 0) then
      print("HTTP request failed")
    else
      print(code, data)
      --node.dsleep(30000000)
    end
  end)
tmr.stop(1) end end)
```

marcelstoer, 2016-02-27:
Again, this works. Try setting the port explicitly i.e. `http://120.26.37.149:80/nodemcu/` for whatever reason it may be required in your environment.

> Why else would the `net` module code work to retrieve data from a local subnet host while `http.get()` doesn't?

Because it connects to port 80 explicitly I figured. If `http.get()` goes to a host in the same subnet as the ESP8286 I also need to set the port in the HTTP URL explicitly.

jfollas, 2016-02-28:
@nonflammable With that static address (192.168.1.200), you get no result for a URL with an IP Address, but you get a result when hitting a DNS name?

Some observations that I made trying to replicate this:
- The ESP does not check to see if an IP address is already in use, and there are no errors generated. I was able to set the same IP on the ESP that my laptop was using, and aside from a connection timeout, there were no other indications that I had set a bad static IP address
- (Probably unrelated) When calling `wifi.sta.setip()`, the DHCP client is shut down. We don't have a way to set a DNS server in this case (i.e, call `espconn_dns_setserver` from the SDK). I suspected this being part of the problem at first, but your scenario is flipped (DNS works, IP does not). Strangely enough, though, DNS queries were still being answered or otherwise resolved - I think because the DNS was previously set by DHCP during the connect, and just wasn't cleared out when `setip()` was called.

Have you tried assigning other static IP addresses on your network to rule out a possible collision with another device?

nonflammable, 2016-02-28:
@jfollas 
> With that static address (192.168.1.200), you get no result for a URL with an IP Address, but you get a result when hitting a DNS name?
Yes

> Have you tried assigning other static IP addresses on your network to rule out a possible collision with another device?
No, because it works with DNS. (changed many times with ESPlorer between DNS/IP)

pjsg, 2016-02-29:
From looking at the code, the http module tries to read the entire response into a single memory buffer. As each packet comes in, it makes new buffer and copies all the old data into the new buffer.

This approach is going to run out of memory pretty quickly if the page is of any significant size. 

What size is your page?

marcelstoer, 2016-02-29:
@pjsg Philip, the issue is, at least for me, that `http://<my-local-subnet-host>/foo.html` doesn't even reach the host but `http://<my-local-subnet-host>:80/foo.html` does. You'd have to sniff WiFi traffic to find out what the differences in the packages sent from ESP8266 are.

However, there's still no conclusive evidence this is actually a bug in our firmware -> I'll close it.

closed 2016-02-29 by marcelstoer