nodemcu/nodemcu-firmware #1090: SSL does not do any server verification -- i.e. provides minimal security

pjsg, 2016-02-27:
It turns out that the default use of the SSL stack on the nodemcu is to not verify any certificates. The actual procedure to verify the certs is a bit gnarly. 

Without doing vertification, the system provides secrecy against people who only have read-only access to the packet stream. If you can intercept the packet stream, then all bets are off. Unfortunately, getting on-path in a wifi environment is easy. 

Yes, it is much easier to create insecure solutions, but, since I do security as my day job, I feel that we ought to make it (at least) possible for people to do the right thing.

jmattsson, 2016-02-27:
I'd love for someone to do the hard yards in the SSL/TLS area. Wishlist includes:

  - Server certificate verification (against user supplied CA or CA list, loaded off SPIFFS)
  - SNI support so NodeMCU can actually talk to more HTTPS servers
  - Ability to use client certficate (again, usersupplied, loaded off SPIFFS)
  - Abilitiy to blacklist chosen cipher(s)
  - Ideally, but it's really too heavyweight imo, CRL checking (or OCSP).

Oh, and DTLS would be great too... I can haz? :smile_cat:

TerryE, 2016-02-27:
Philip, all this says to me is that the current ESP generation isn't yet ready to be used as a secure device-to-user capable component, but was already my view.

jmattsson, 2016-02-27:
No, some of this is NodeMCU's fault - we should be exposing more stuff so SSL *can* be used better. At least that's my understanding, I haven't poked in this area for a bit.

marcelstoer, 2016-02-27:
> It turns out that the default use of the SSL stack on the nodemcu is to not verify any certificates. 

Yep, I found out last night by coincidence. I tested #1089 and at some point had a typo in the URL so instead of http://120.26.37.149/nodemcu/ I was connecting to http**s**://120.26.37.149/nodemcu/. Apart from the fact that the returned content is different it surprised me that NodeMCU didn't complain about the invalid cert.

btsimonh, 2016-03-01:
In mqtt, client side cert can be used (and may be required for Amazon IOT?).  If we're looking at TLS and certs, then including mqtt/TLS in the review would be good.

pjsg, 2016-03-02:
My thoughts are to do the simple stuff that is supported by the espressif sdk. This would include:

* Verify server certificates using a local (small) trust store. The SDK supports two certs.
* Use a client certificate when making a connection. 

From my reading of the docs, it appears that there is no onboard key generation to support client certificates. This means that the key has to be injected as part of the bring up of the platform. You don't want to burn the private key into the code (unless you are only making one instance). You should never reuse private keys between devices.

The Espressif API is a bit baroque and requires that the keys/certs etc be in the flash and aligned on a 4k boundary. This is a pain! 

Thoughts on API:

* Allow user to specify up to two certs at firmware build time to be used for verifying servers. These would get included into the bin file.
* Allow a user to specify the certificates at run time. This would flash the certs into the spare page and then use them.
* Allow the user to specify the client cert/private key at run time. This would flash the certs into another spare page of flash.

Not sure that I see the need to support more than one set of certs at a time (i.e. using multiple pages).

If you specify MODULES_SSL, then I would default allocate the two extra (4k) pages. There will be options to disable the client side certificate page -- but I really want to encourage people to think about verifying server certs.

Unfortunately, I can't see a way to get these pages from SPIFFS -- it has already subdivided all the pages and put data structures there (as far as I can tell). 

Thoughts?

jmattsson, 2016-03-02:
4k page alignment requirement for certs?! What were they (not) thinking?? Or does it maybe only apply if you want to be able to "simply" install the certs afterwards? Doing a single page-write is considerably simpler than doing a read-modify-write cycle. I'd suggest testing whether that 4k requirement is really a requirement - I'd be surprised if it can't use certs from RAM as well.

If it does need to be on-4k-in-flash, then the "easy" way would be to do a `const char certX[SPI_FLASH_SEC_SIZE] __attribute__((align(SPI_FLASH_SEC_SIZE)))`. If that yields too much linker fill, then dropping those cert pages into its own section would be an option.

pjsg, 2016-03-02:
The API that they provide is that you pass in the *page* number in flash where the certs are stored. They then access the certs directly. What isn't clear to me is how this maps into the address space -- where does the flash actually start?

jmattsson, 2016-03-02:
...
...
... That's *horrible* API design! :(

You'd effectively be looking at something like this then: `platform_flash_mapped2phys(certX_addr) / SPI_FLASH_SEC_SIZE)`.

pjsg, 2016-03-03:
The good news is that I have a (very flaky) version of the code that does client side certificate verification! I suspect that some enhancements from espressif will make it much more usable.

pjsg, 2016-03-06:
It looks to me as though the chances of making client certificate authentication work are small at the moment. I think that some improvements need to be made to the SDK. 

I'll probably submit the current code (for server verification) and leave the client auth code disabled.

pjsg, 2016-03-07:
Submitted PR #1125

pjsg, 2016-03-19:
The original PR was merged. However, the client side stuff still doesn't work.... Waiting on espressif.

djphoenix, 2017-01-23:
I think this may be closed because now we are thinking about #1700.

closed 2017-01-23 by djphoenix