nodemcu/nodemcu-firmware #1092: File module review and reimplementation

hvegh, 2016-02-27:
File module seams to allow only access to one file at a time..

What do you think maybe use pattern like?:

> fh = open(...)
> ..
> read(fh)
> write(fh, ..)
> close(fh)

devyte, 2016-02-27:
In that case, wouldn't it be better to try to stick with the lua standard, and go with something like io.open(...) and so on?

TerryE, 2016-02-27:
Can we have a sanity check here?  Abstract use patterns or one thing, but we are talking about two bits of silicon and a wifi aerial on a tiny module -- a 1st generation programmable IoT device.  It's little short of a miracle than we have _any_ R/W file system.  Having one which supports multiple files open will be a step increase in resources needed to serve the FS.   Maybe on the ESP32, eh?

jmattsson, 2016-02-28:
I was initially after the very same thing, and even started writing the code for it, but while digging into SPIFFS and the `file` module I reached the conclusion that the necessary backing buffers would be too much on the ESP8266. Right now we have SPIFFS tuned with the minimal buffers it can work with, which saves us a ton of RAM. It comes at the cost of not being able to handle one open file at a time (okay, technically it's two, one for a `file.open()` and one for `dofile()`).

What I ended up doing was simply stashing the read offset for each file in a Lua variable and reopen as/when necessary. It's a little bit more runtime overhead, but we have a lot more CPU cycles than we have RAM.

TL;DR: Unless someone can come up with something exceptionally clever to not increase the backing buffers, this will do more harm than good.

hvegh, 2016-02-28:
Valid points. The use case here is an ESP as a simple stand alone web server with a few static html pages stored as files in flash. 

Just thinking, when a file is opened read only, there would probably be no need for a backing store, data could probably be directly served from flash, independent of the number of readers.

So under the restriction of one file written or modified at a time globally you could allow read only access to the other files without the additional memory for backing store. Of course per open file you would need some bytes: 8 bytes for linked list, 4 bytes for a pointer to the file descriptor, 4 bytes for an file offset and probably 4 bytes for dev abstraction...

@devyte: io.open read write seams a good idea.  
It could be neatly abstracted behind io.open() io.read() io.close()

Of course once you try to modify a file with active readers ... you would get an open error..
Once you attempt to create or modify a file when there's already a file open for writing you would also get an open error.. 

@jmattsson: The reopen while keeping track of the offsets seams pretty good idea.

TerryE, 2016-02-28:
@hveng Henk this isn't one to "think about": two of the main committers have given you the same answer: we arne't going to do this because the RAM impact is too high.  If you are serving a muber of pages through separate sockets then you are already pushing the system resources.  Since the TCP doesn't do packet marshalling, you need to work in maxIP packet chunks.  Just do as Johny suggests and reopen each file to read one or more vhunks at a time and note the file pointers to reset them on the next read.  

I really can't understand who you think we would consider a complete rewrite of the file system to avoid a few dozen extra lines of Lua at an application level.

closed 2016-02-28 by TerryE