nodemcu/nodemcu-firmware #1093: Extending the GPIO driver to include hook support

TerryE, 2016-02-27:
I have pulled this discussion into a separate issue because we have two quite separate points: 

-   If and how we include a ISR level hook mechanism in the GPIO driver
-   The implementation of a Rotary Switch module (which uses this hook) which is covered in Issue #1058 and the PR  #1067.

This issue is to enable discussion on this more general platform wide point.  @pjsg Philip has proposed a a new API call within the platform layer to enable other drivers to declare a hook and extra code within the GPIO ISR to enumerate any established hooks.  This concrete example helps us to focus discussions. 

At a functional level, my view largely parallels Philip's but with some variances:

-    The platform GPIO interface should support a hook mechanism.
-    A hook should be able to operate across multiple IO pins.
-    Other drivers should be able to register _and unregister_  such hooks.
-    The names of these functions should not be called "callbacks"  but something more meaningful, e.g. <br>
`  platform_gpio_register_intr_hook(ISRhook_name, ISRhook_mask);`<br>
`  platform_gpio_unregister_intr_hook(ISRhook_mask);`
-   Since this is global ISR overhead, some though should go into making any overhead minimal.
-   Hooks should be able to suppress the normal  task post processing.

So some implementation notes: the current proposal is:
```C
  GPIO_HOOK *hooks = gpio_hooks;
  for (; hooks; hooks = hooks->next) {
    if (gpio_status & hooks->gpio_bits) {
      hooks->callback(gpio_status & hooks->gpio_bits);
    }
  }
  // ... scan of the gpio_bits to determine which to action. 
 ```
-   I would prefer to maintain a intersection of all hook masks as `gpio_hook_bits`and stick to simple linear arrays rather than a linked structure for ISRs:
```C
  GPIO_HOOK *hooks = gpio_hooks;
  if (gpio_status & gpio_all_hook_bits) {
  for (j = 0; gpio_hook_bits[j]; j++) {
     if (gpio_status & gpio_hook_bits[j])
       gpio_hook_function[j](&gpio_status);
  }
```
-   This has a 3 instruction overhead if the changed bits do not relate to a hook,
-    If they do relate to a hook then the per hook overhead is smaller
-    The gpio_status is passed by reference to allow the hook routine to do all processing related to
the pins and clear down the corresponding bits in the status to prevent the main loop posting a callback task (as the hook can also do this if necessary).
-    The register and unregister need to be atomic wrt the ISR, and carry out the necessary bit conflicts (e.g. two hooks cannot claim control of the overlapping bits.)  The current code doesn't do this.

Philip, do you differ from this?  Do you want me to propose a mod to your rotary patch?  Anyone else have views on this?

I'll hang fire on commenting on the other PR until we've gone around this once.

pjsg, 2016-02-27:
Thinking about this some more -- I think that we probably ought to move the existing GPIO interrupt handler and make it a customer of the primary hooking-only handler. This way it becomes simpler to make sure that we aren't even routing the same interrupt to more than one handler. Also, it probably becomes cheaper to handle an interrupt as we can early exit as soon the interrupts have been processed (and don't need to iterate over all the potential triggers in the gpio code)

In practice, I would expect the non-gpiomodule interrupts to fire in preference to those routed to the gpio module. This is purely because the reason that you use the hooking approach is because you are much more latency sensitive. 

I think that it is worth getting this right.....

pjsg, 2016-02-29:
I looked at trying to make the existing code use the same approach, but it got messy quite quickly when trying to deal with GPIO_INTERRUPT_HOOK_ENABLE being enabled and disabled.

I can try and work on it some more if you think it worthwhile.

TerryE, 2016-02-29:
Philip, if you code depends on `GPIO_INTERRUPT_HOOK_ENABLE` then put an `#ifndef` guarded `#error` on this. This idea of this configure define is to avoid the memory overhead if your apps don't need it, but if they do then they do. You approach is the best and most robust for this use.

closed 2016-03-01 by TerryE