nodemcu/nodemcu-firmware #1094: new BMP085 Module always the same result

willwach, 2016-02-28:
I'm trying to use the BMP085-Module from the following page:
http://nodemcu.readthedocs.org/en/dev/en/modules/bmp085/

sda on 1, scl on2

My code:
function readBmp(sdaPin, sclPin)
    bmp085.init(sdaPin, sclPin)
        
    local t = bmp085.temperature(1)
    print(string.format("Temperature: %s.%s degrees C", t / 10, t % 10))
    
    local p = bmp085.pressure(1)
    print(string.format("Pressure: %s.%s mbar", p / 100, p % 100))
end

It always leads to:
Temperature: -13.2 degrees C
Pressure: -306.27 mbar

If I remove the BMP085 Module, its the same.
What am I doing wrong?

Or is this the also the result if the module is broken?

Would be nice if someone could help me.

dvv, 2016-02-28:
Seems like I2C gives constant 0xFF values. Had such problem with one module.

willwach, 2016-02-29:
Could you tell me, how the module should react if nothing is connected to i2c?
Maybe its the default behavier if nothing is connected to answer with these values?

TerryE, 2016-02-29:
See #719

closed 2016-02-29 by TerryE