nodemcu/nodemcu-firmware #1095: Shouldn't the WiFi module allow you to set nameservers?

marcelstoer, 2016-02-28:
I reflected a bit on #650 (after I answered it).

Isn't it odd that the WiFi module doesn't allow you to set nameservers, see http://nodemcu.readthedocs.org/en/dev/en/modules/wifi/#wifistasetip. Instead there's [`net.dns.setdnsserver`](http://nodemcu.readthedocs.org/en/dev/en/modules/net/#netdnssetdnsserver) but that made me wonder:
- Does `net.dns.setdnsserver` only influence network operations triggered from the net module itself? That'd be hard to believe.
- What if you excluded the net module but include the WiFi module
- Shouldn't `wifi.ap.setip` be renamed? It allows you to set 3 network-related parameters one of which is the IP address.

jfollas, 2016-02-29:
I was expecting `setdns` to be part of `wifi` when I looked for it earlier, since that's the module that controls our interface to the network. Initially, I found it acceptable to be part of `net` (because at least there's a way to set DNS if you use static IP addressing). But, now that you mention the possibility of the `net` module not being included, it has me rethinking.

I personally think `setip` is descriptive enough. `wifi.{IF}.setip` is a wrapper around the SDK function `wiﬁ_set_ip_info`, and maybe Espressif's additional of "info" makes theirs a better name. But "IP" by itself does not mean just the address (you need all three parts to configure IP).

closed 2016-08-07 by marcelstoer

enaon, 2017-08-02:
I had the same problem as rubberbird. 

When using the esp in STATIONAP mode as router or napt device, it is useful to be able to offer a valid dns server to the clients on the AP interface, and not the esp itself. 

I took a go at it, I named it net.dns.setapdnsserver(dns_ip_addr) to follow the sta dns setup, I don't really know how to code but it works. 

https://github.com/enaon/nodemcu-firmware-router/tree/1.5.4.1-final-ipfwd-apdns

marcelstoer, 2017-08-02:
Who is rubberbird? Which problem?

I initially created this issue because I was expecting something like `wifi.dns.setdnsserver` rather than `net.dns.setdnsserver`. Hence, it's only partially related to your changes. However, I see how your changes could be useful. If you consider https://github.com/nodemcu/nodemcu-firmware/blob/master/CONTRIBUTING.md and made a PR against the dev branch we will consider it.

Btw, in the meantime I've understood and accepted why `dns.setdnsserver` is with the net module rather than with wifi. It just means that the net module really _is_ essential for any NodeMCU firmware.

enaon, 2017-08-02:
rubberbird is the guy that started this [issue](https://github.com/nodemcu/nodemcu-firmware/issues/650) that led to the one we are now are.

the problem with net.dns.setdnsserver is that it concerns only the esp dns needs. If that esp is in STATIONAP mode with ip forward enabled (and maybe nat), the clients that connect to that same esp are using the ipaddr of the ap if of the esp,  they can't resolve names and can't connect to the internet. 

This is solved with something like net.dns.setapdnsserver, so that ap clients get an offer for a valid dns server. 

I took a second go and now I have napt enabled, works ok, I now have to enable forwarding. 
If this seems good for general use, please guide me as how to proceed.

https://github.com/enaon/nodemcu-firmware-router/tree/1.5.4.1-final-napt-ipfwd-apdnsthanks a lot.