nodemcu/nodemcu-firmware #1096: MQTT Publish Limitations

btsimonh, 2016-02-28:
If using MQTT Client, be aware that messages only seem to leave NodeMcu once per second, and if you post >10 or so, the module may crash (i.e. don't try to direct LUA output there....).

pjsg, 2016-02-29:
I believe that this is resolved in the dev branch. I committed a fix to this several weeks ago.

btsimonh, 2016-02-29:
i'm testing on:
NodeMCU custom build by frightanic.com
	branch: dev
	commit: 716e6824db422aab19ae22419fec1d2a8990b9e3
	SSL: true
	modules: adc,file,gpio,http,mqtt,net,node,ow,pwm,rc,sntp,tmr,uart,wifi
 build 	built on: 2016-02-21 17:54

I see your pull comitted 19 days ago, and i think the above should include it.  What I need to check is what QOS i'm asking for, as you highlight that QOS=0 seems much improved, but other values may have implications.

pjsg, 2016-03-07:
The two files actually contain different parts of the code. I think that I have identified the issue with qos~=0 . Can you post a small sample of code that doesn't work for you -- so I can verify the fix.

Thanks

pjsg, 2016-03-19:
The other MQTT fix was merged that dealt with qos ~= 0.

If I don't hear that this issue is still present in the latest dev, I'll close this....

pjsg, 2016-03-21:
Asserting this issue is closed.

closed 2016-03-21 by pjsg