nodemcu/nodemcu-firmware #1098: Problems reading DS18B20

fotastisch, 2016-02-29:
Hi there!

I'm running into problems reading the DS18B20. I think I partly solved some of my problems by switching to the nodemcu dev branch. Since the switch my nodemcu board shows the number of attached DS18B20s correctly as 1 instead of 0. But the temperature readings still don't make any sense (to me): Supposedly the ambient air temperature is something between 80 and 90 °C. This is approx. 4 times too high, I expect something of approx. 20 °C. I have no idea what the problem is. I'm using the code and example from the DS18B20 module.

Any ideas are appreciated!

btsimonh, 2016-02-29:
This is working for me.
I did have the same issue as you, but can't remember what resolved it exactly; something to do with moving to dev branch (there was some very specific reason for getting 85C - it's the default value if a read was not triggered?).
Do make sure you have a pullup resistor - mine worked up to 3 devices without (even triggering all at the same time), but with 4k7 pullup and triggering one after the other, I now read 5. 
Not the best code because it waits 1s in the loop; this should really be asynchronous; my next job.

https://github.com/btsimonh/NodeMCU-temp-to-ThingSpeak/blob/master/ds1820.lua

s

fotastisch, 2016-02-29:
Hi!

Thanks for your quick reply! I'll try your code asap.
I don't receive 85 °C exactly, but varying values. I can see them increase and decrease when they should, but the values remain rigged.

Oh, and the 4.7k pull up is in place. When I've tried your code I'll post my status. ;)

TerryE, 2016-02-29:
This issue read more like a thread that belongs on esp826.com.  Please see our [support policy](http://nodemcu.readthedocs.org/en/dev/en/support/) and Issue #719.  If you determine that there is a specific bug with the firmware or this library then please provide the details and reopen this issue.

closed 2016-02-29 by TerryE